janelia-arduino/TMC2209 #1: Do you plan to add support for vactual

angelnu, 2021-12-26:
I just found your library in Platformio and since it is specific for the TMC2209 I started with it instead of the larger TMCStepper library. Thanks a lot for sharing!

One thing I noticed that is missing that I would need is the ability to set VACTUAL so the stepper can be moved using the serial port instead of STEP/DIR. Would you be willing to add that? An example of what I mean is in [this Gist](https://gist.github.com/metalinspired/dcfe07ed0b9f42870eb54dcf8e29c126)

I could also try a PR as soon as I fix another issue - as soon as I use your library the stepper warns up quite a lot (not burning but hot) which does not happen when I drive the stepper via DIR/STEP. I tried with enableAutomaticCurrentScaling() but it does not make anything: stepper_driver.getSettings(),automatic_current_scaling_enabled stays at zero.

I will continue tomorrow.

peterpolidoro, 2021-12-26:
Hi. Sure I can add support for vactual. I am taking a few days off for the holidays, but I will add that as soon as I get back.

When your motor gets hot, can you reduce the motor run and hold current values with the library commands to see if it runs at a lower temperature?

peterpolidoro, 2022-01-05:
I added some methods so you can try to set vactual. I do not have it attached to a motor at the moment so I cannot test it. Can you please try it when you have a chance and let me know if it works? Thanks!

peterpolidoro, 2022-01-21:
I added the methods:
`void moveAtVelocity(int32_t microsteps_per_period);`
`void moveUsingStepDirInterface();`

moveAtVelocity uses vactual to spin the motor with no step and direction signals necessary.
moveUsingStepDirInterface disables vactual so the step and direction signals control the motion.

I have tested the methods and they seem to work fine.

closed 2022-01-21 by peterpolidoro

Fr33man85, 2022-04-01:
hi,
can you explain - in short - , how to generate a certain amount of steps with moveAtVelocity?

I want to send something like "do 5 steps forward" through uart, but i can not figure out, how to do that..
I tried to read the datasheet of the trinamic 2209, but i dont unterstand the vactual formula in it.

i belive, that iam only "to dumb" right now, because as far as i know, marlin/klipper also communicate over uart with the stepper driver of an 3D-printer. so a precice way to send a certain amount of steps is needed and surely possible.

thank you in advance.

PS: is there an detailed documentation about all your functions in this library?

aagum-bae, 2024-05-16:
Hi I am looking to find solution for the same, did you find a way to tell the motor to move x steps using only UART?