janelia-arduino/TMC2209 #3: Issue compiling for any ESP32 board

aramazhari, 2022-01-22:
Hello,

I noticed the define DISABLED is already being used by the ESP32 library and this creates a conflict with the TMC2209.h's declaration with the same name:

```
In file included from Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/esp32-hal.h:53:0,
                 from Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:35,
                 from sketch\TestConnection.ino.cpp:1:
Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/esp32-hal-gpio.h:51:19: error: expected unqualified-id before numeric constant
 #define DISABLED  0x00
                   ^
Arduino\libraries\TMC2209\src/TMC2209.h:169:24: note: in expansion of macro 'DISABLED'
   const static uint8_t DISABLED = 0;

```

Since ESP32 is a widely used board with a standard library, I suggest renaming TMC2209's DISABLED and ENABLED to something more unique to avoid the compilation error.

Referring to this line:
https://github.com/janelia-arduino/TMC2209/blob/bf8d6efcb4b8359e9494021a9917dd26d192c0a8/src/TMC2209.h#L169

Thanks,
Aram

peterpolidoro, 2022-01-22:
Oh thank you for finding this issue and telling me! I will change it as soon as I get the chance.

peterpolidoro, 2022-01-24:
I renamed ENABLED and DISABLED. Can you please try it now and see if it fixes your problem?

aramazhari, 2022-01-24:
Hi Peter,

I tried and it works. Thanks a lot!

closed 2022-01-25 by peterpolidoro