janelia-arduino/TMC2209 #6: Communications but no setup

ita1024, 2022-04-24:
While the communication would succeed, the following modifications are necessary to have the examples run locally (version 8.0.5) from a Trinket M0:
- Add `delay(10)` at the start of `TMC2209::read()`
- Add `delay(10)` at the end of `TMC2209::read()` before the return statement
- Add `delay(10)` at the start of `TMC2209::write()`
- Add `delay(10)` at the end of `TMC2209::read()` after `sendDatagram`
- Call `setRunCurrent(75)`
Changing the baud rates did not improve the situation (it works with the default value).

Additionally, the following observations were made while investigating the above:
- The method `read()` should checks if the Datagram's CRC is correct
- The method `write()` should check if the operation succeeds: adding `getInterfaceTransmissionCounter();` calls before and after the write operation helped determine that the counter was incremented only after adding the delay described above
- The method `write()` is returning void, so there is no way for a library user to handle errors
- A method for setting the current in mA would be more adequate than percentages, or perhaps `setRunCurrent` should be named `setCurrentPercentage`
- It is unclear from the API how one would rotate the stepper by a number of steps and if that would require additional cables or if this can be achieved from the API
- For consistency, `serialOperationMode` would be best called `isSerialOperationMode
-  The attribute `TMC2209::Input::pdn_serial` seems to be called `PDN_UART` in the datasheet

peterpolidoro, 2022-04-25:
You make some good points here, thanks!

I was hoping to avoid adding more delays since those can really hurt real-time performance. Is 10ms really necessary in each of those cases? Did you try using smaller delays to see if those would work as well?

What do you think read and write should do if they detect an error? Return some sort of error value?

Calling getInterfaceTransmissionCounter() twice for each write would add many serial transmissions. With the delays in the serial transmissions, those could really decrease performance to catch the very occasional write error. The user could always add those checks to their code by calling getInterfaceTransmissionCounter() so they could decide whether or not error detection is worth the performance decrease.

It would be nice to be able to use mA for units rather than percentages, but the firmware cannot always make that conversion as it does not know the motor supply voltage or the sense resistor values. You make a good point, though, that perhaps the methods should be renamed.

From this API, you cannot easily rotate the stepper by a specific number of steps. As I attempted to explain in the README (please tell me if I need to improve the documentation somehow) a separate stepper controller is necessary to send step pulses to the driver to make it step a specific number of steps. This driver has an internal step generator and it allows you to move the motor at a specific step rate. So if you set that rate for a given amount of time you can in effect set the specific number of steps, but delays in the system might cause the actual step count to be off slightly.

Good point about renaming serialOperationMode.

I used the term "serial" instead of "uart" in an attempt to be more consistent with Arduino naming, although you are right, that makes it more inconsistent with the names in the manual. Perhaps I should rename that input to make it consistent with the manual, I will think about it.

Thanks!

peterpolidoro, 2022-04-25:
> After changing delay(...) to delayMicroseconds(...) the lowest value found is 110 microseconds.

So every one of the delays that you listed in your first message are necessary and a value of 110 microseconds works for every one of those delays?

> A class attribute containing the last error would be precious for troubleshooting, and would not over-complicate the API. Why is the driver not communicating? How to recover from it?

I will work on adding the error checking when I have a chance. I am also happy to look at pull requests if you have ideas for how to implement this.

> How can one call TMC2209::getInterfaceTransmissionCounter to check for errors before calling TMC2209::setup? The method TMC2209::setup seems to be setting the baud rate and writing to several registers

So you are looking for more error checking and debug information to help you diagnose connection problems before running setup?

> The description on Github reads 

Yes good point, the description is a little confusing since the name of the library is the same as the name of the chip. I should change the description to be about the library instead of being about the chip.

ita1024, 2022-04-25:
> So every one of the delays that you listed in your first message are necessary and a value of 110 microseconds works for every one of those delays?

I have not tried removing any the delays yet.

> So you are looking for more error checking and debug information to help you diagnose connection problems before running setup?

No, this was an example to show why `getInterfaceTransmissionCounter` checks should really be in `TMC2209::write` and why this would not have helped with troubleshooting with this bug report. In the same train of thought, one could add delays in user code, but this fails for the `TMC2209::setup()` method.

One could think of writing a subclass but the methods `TMC2209::write` and `TMC2209::read` are private.

peterpolidoro, 2023-04-28:
I have just released a new version 9.0.0. When you have a chance, please test the new changes and see if the new version resolves any issues you may be having with the previous version. If you find any bugs or would like to request any new features please leave me a comment or open a new issue. Thank you!

closed 2023-06-27 by peterpolidoro