janelia-arduino/TMC2209 #7: am i doing something wrong?

EinRainerZufall, 2022-05-02:
I can't get a connection, but with the TMCstepper lib it works fine.

I get this message from the TestSetup example:
```
Stepper driver is not communicating!
Try turning driver power on to see what happens.
```

My wiring diagram:
[image]

My program:
[TestSetup.txt](https://github.com/janelia-arduino/TMC2209/files/8603872/TestSetup.txt)

peterpolidoro, 2022-05-02:
At a quick glance it looks like you are doing everything correctly. Is it possible for you to look at the UART signal on an oscilloscope? Another user pointed out that I may need to add more delays to make the communication work properly with some boards. I do not have access to an ESP32S-Hiletgo board so I cannot test this directly, but I can let you know after I have made some modifications so you can test it again if you would like.

EinRainerZufall, 2022-05-02:
Oh sorry i using a noname ESP32 Dev board. Unfortunately i haven`t a oscilloscope but the i checked the Serial2 with a TTL to USB adapter. And the same wiring works fine with the TMCstepper LIB. Yes, that would be nice if you could let me know when you have made the modifications

Mattstone99, 2022-06-11:
Hi, I was using an Arduino Mega with the same code, but I can't seem to communicate with the driver. Do you have a test circuit I could reference?

kgallimore, 2022-08-20:
> Hi, I was using an Arduino Mega with the same code, but I can't seem to communicate with the driver. Do you have a test circuit I could reference?

I managed to get the TMCStepper Library to work on the Mega, but for some reason I can't establish a connection with this library.

tomasz90, 2022-12-24:
Same here tested on ESP32 DEV KIT, librarty doe not work...

closed 2023-06-27 by peterpolidoro