janelia-arduino/TMC2209 #9: Adding support to Software defined serial ports

eliocor, 2022-08-29:
Do you think it will be possible to add support (with the proper limitations) to software defined serial ports like in the "SoftwareSerial" library: [https://docs.arduino.cc/learn/built-in-libraries/software-serial](https://docs.arduino.cc/learn/built-in-libraries/software-serial) ?
Just as an example, the [TMCstepper](https://github.com/teemuatlut/TMCStepper) library works flawlessly with SoftwareSerial ports.

.        TIA
_          Elio.

P.S.: always available to test the code.

peterpolidoro, 2022-08-31:
Hi. I can take a look, but it might take me a few weeks. I do not have the hardware setup to test changes to the code at the moment. I am happy to look at a pull request, though, if you would like to make some changes and test it on your setup before I have a chance. Thanks!

peterpolidoro, 2022-09-23:
Have you tested the TMCStepper library with a TMC2209 using SoftwareSerial? I am wondering if there will be any problems with SoftwareSerial not being able to simultaneously transmit and receive. The TMC2209 unfortunately has a single wire UART, which makes communication a little tricky. In my library I have to throw away all of the data that I transmit to the TMC2209 but gets read back into the microcontroller because the transmit and receive lines are connected. If the TMCStepper library does indeed work with SoftwareSerial using the TMC2209, I will have to dig into the details of how they are handling the connected transmit and receive lines. I could just modify the library and you can try and see if it still works, but I am guessing their could be problems unless that is figured out correctly.

eliocor, 2022-10-07:
sorry for the late reply....
Yes, as I previously wrote, I tested it and it works flawlessly.
Here is a code sample (reduced to its minimum terms) you can use as a reference: no STEP & DIR pins are defined, only direct commands to the TMC2209.

`#include <TMCStepper.h>         // TMCstepper - https://github.com/teemuatlut/TMCStepper
#include <SoftwareSerial.h>     // Software serial for the UART to TMC2209 - https://www.arduino.cc/en/Reference/softwareSerial

#define DRIVER_ADDRESS   0x00   // TMC2209 Driver address according to MS1 and MS2
#define EN_PIN           8      // Enable - PURPLE
#define SW_TX            A5      // SoftwareSerial receive pin - BROWN
#define SW_RX            A4      // SoftwareSerial transmit pin - YELLOW
#define R_SENSE 0.11f           // SilentStepStick series use 0.11 ...and so does my fysetc TMC2209 (?)

SoftwareSerial SoftSerial(SW_RX, SW_TX);                          // Be sure to connect RX to TX and TX to RX between both devices
TMC2209Stepper TMCdriverX(&SoftSerial, R_SENSE, DRIVER_ADDRESS);   // Create TMC driver

void setup() {
  SoftSerial.begin(115200);           // initialize software serial for UART motor control
  TMCdriverX.beginSerial(115200);      // Initialize UART
  
  pinMode(EN_PIN, OUTPUT);           // Set pinmodes
  digitalWrite(EN_PIN, LOW);         // Enable TMC2209 board  

  TMCdriverX.begin();                                                                                                                                                                                                                                                                                                                            // UART: Init SW UART (if selected) with default 115200 baudrate
  TMCdriverX.toff(5);                 // Enables driver in software
  TMCdriverX.rms_current(500);        // Set motor RMS current
  TMCdriverX.microsteps(64);         // Set microsteps

  TMCdriverX.en_spreadCycle(false);
  TMCdriverX.pwm_autoscale(true);     // Needed for stealthChop
  TMCdriverX.VACTUAL(50000);            // Set motor speed  
}

void loop() {
  static bool dir = false;

  delay(2000);
  dir = !dir; // REVERSE DIRECTION
  TMCdriverX.shaft(dir); // SET DIRECTION
}
`

peterpolidoro, 2022-10-07:
Oh interesting, yes I would love to get this library to work with SoftwareSerial then. I will make an experimental branch to test it.

eliocor, 2022-10-07:
Always available in being a beta tester

peterpolidoro, 2022-10-10:
Would you consider [AltSoftSerial](https://www.pjrc.com/teensy/td_libs_AltSoftSerial.html) instead of SoftwareSerial?

eliocor, 2022-10-11:
I think using *SoftwareSerial*
<https://docs.arduino.cc/learn/built-in-libraries/software-serial> should
be better:
- it is a built-in library, so it will not require to load "by hand"
another library
- it will solve your problem to filter out the characters you are sending
to the TMC220x [...It cannot transmit and receive data at the same time....]

raulperez, 2022-10-17:
I am also interested on the SoftwareSerial. Is there any code that can be tested? :)

peterpolidoro, 2022-10-17:
Thank you all for your interest, I will definitely add SoftwareSerial support as soon as I have a chance and when I figure out how best to implement it.

The best, from a firmware viewpoint, would be to add hardware to the circuit to convert a single wire UART into a traditional two wire UART. Not all circuits will have that option, however, so I think I need to support both single and dual wire UART possibilities, both for hardware serial and software serial.

I am not sure how to implement the proper timing for the single wire UART without adding ugly software delays.

I have to research this some more before I can complete it. I may have a chance to work on this in about a week.

GuusvanMarle, 2023-03-23:
Hello all,

I was just wondering if there are any updates on this.

best regards,
Guus

peterpolidoro, 2023-03-23:
I will be working on this in the next couple of weeks.

Neumi, 2023-03-26:
@peterpolidoro I built a TMC2209 Arduino UNO test shield with all pins connected to DIOs, and Software Serial on pins 9 and 10. If you like, I can send you one. https://github.com/Neumi/tmc2209_devboard

peterpolidoro, 2023-04-28:
I have just released a new version 9.0.0. When you have a chance, please test the new changes and see if the new version resolves any issues you may be having with the previous version. If you find any bugs or would like to request any new features please leave me a comment or open a new issue. Thank you!

Neumi, 2023-05-02:
I've just tested [SoftwareSerial.ino](https://github.com/janelia-arduino/TMC2209/blob/main/examples/UnidirectionalCommunication/SoftwareSerial/SoftwareSerial.ino) on my [TMC2209 devboard](https://github.com/Neumi/tmc2209_devboard) and it works flawlessly with software serial!
I would consider this issue solved. Thanks! :)

closed 2023-05-02 by peterpolidoro