janelia-arduino/TMC2209 #11: "Stepper driver not setup and communicating!"

musapinar, 2022-10-28:
Hello,

This is not exactly related to this library but search engines failed on helping me.
I'm using an Arduino Mega Pro 2560 and a TMC2209 V2.0 from Makerbase.
I uploaded the "StallGuard.ino" sketch example and my Serial monitor outputs this error continuously.

Here is my wiring:
Mega is powered via 5V USB.
I make a 5V/GND rail from the 5V output of the Mega.
That 5V rail powers the TMC2209, via the VDD pin
EN pin of the TMC is connected to GND
1B/1A/2A/2B on the TMC are connected to a NEMA17
GND and VMOT on the TMC are connected to 12V from my bench PSU.
Pin labelled "UART" on the TMC is connected to pin D15 of the Mega (which is Serial3, so I modify the sketch accordingly)
Pin D16 is connected, through a 1K resistor, to D15.

What am I missing?
I tried with other hardware serials (Serial1 and Serial2) without any success.
I tried switching RX/TX pins, maybe, for each hardware serials, without any success.
I tried with an other TMC chip, without any success (I have 5 of those).

I am out of clues now.
I can see on the TMC chip that the "UART" pin is connected to the pin next to it, through R8, which is a zero ohm resistor. Am I expected to remove it (0201 size i'd say)? 
Thank you.

musapinar, 2022-10-29:
[image]

peterpolidoro, 2022-11-02:
Hi. I am just now looking at this. At a quick glance I do not see any obvious errors in your wiring. The  [Makerbase board schematic](https://github.com/makerbase-mks/MKS-StepStick-Driver/blob/master/MKS%20TMC2209/MKS%20TMC2209%20V2.0_001/MKS%20TMC2209%20V2.0_001%20SCH.pdf) very similar to the [SilentStepStick TMC2209 schematic](https://www.trinamic.com/fileadmin/assets/Products/Eval_Drawings/TMC2209SilentStepStick_schematic_v20.pdf) so you probably do not need to modify the board.

I will see if I can find a Mega board so I can test it too.

Have you tried the [TMCStepper](https://github.com/teemuatlut/TMCStepper) library? Does that work?

musapinar, 2022-11-03:
I will try with other kinds of boards (Nano, SAMD21) later this week-end. I'm aware of that library and I do not see any reason for it to work any better, but I will give it a shot as well.
Makerbase replied to my email. They don't know much about Arduinos. They are focused on 3D printers. These PCB are supposed to fit in Dupond slots on a steppers controller board I believe, where one is supposed to move a jumper to enable UART mode on/off.
Thank you for your help.
I will keep you updated.

musapinar, 2022-11-05:
That other library worked but only because it drives the motor through the STEP/DIR pins. No success with UART. It must be the chip.

peterpolidoro, 2022-11-07:
I will test it with a Mega board, there must be a way we can get the UART to work. Do you have an oscilloscope? You could look at the UART line and see if something funny is happening.

peterpolidoro, 2022-11-07:
I just tested the library with a Mega 2560 Rev 3 and it seems to work fine. I included a link on the README to the schematic of the wiring I used to test it. I used the 5V output of the Mega for the Vm input on the TMC2209 SilentStepStick and I used the 3.3V output of the Mega for the Vio input. The TMC2209 datasheet says the Vio input voltage range is from 3-5.25V. Have you checked the 5V output? Maybe it is slightly higher than 5.25V.

musapinar, 2022-11-08:
No, I do not have an oscilloscope.
Yes, I have a constant 5.00V. The board has two LDOs (AMS1117), one for 3.3V, one for 5V. It's reliable.

I tried with a standard Mega (non "pro") with a wiring close to yours: no success.
It would have been great if MakerBase was able to tell me what exactly the jumper enabling UART on their 3d printers does.
I've tried an absurd amount of combinations and I'm confident the problem is the MakerBase PCB.
I've ordered a PCB with a pinout similar to your "SilentStepStick".

You tried your best in helping me. Thank you very much for your time.
Feel free to close this issue. I'll post the end of the story in a few weeks, when the pcb has arrived, if it's still open.

beamsjr, 2022-12-08:
I am having the same issue, unable to get UART to work, I am using an ESP32 with a TMC2209 v1.2.  There have been a few times that the motor would start to move using uart, but always getting Stepper driver not setup and communicating!  I have hooked it up to the PC using salie logic to see it the uart was producing, and will attach a screenshot.  It looks like at least something is trying to talk,  unfortunately I don't know how to troubleshoot anymore with the signals.  I would love to try and dig in deeper if you have any ideas. Thanks, Joe

[image]

beamsjr, 2022-12-08:
I replaced the 1k resistor with a 10k ohm resistor and it is working great.  Also I have my PSU ground connected to my ESP32 ground, I have read that they should be connected, any thoughts? I noticed @musapinar did not have his connected on the schematic.

peterpolidoro, 2022-12-08:
@beamsjr very interesting, thank you! @musapinar when you have a chance can you please trying using a 10k resistor instead of a 1k? If that works I will update the documentation. If the grounds are not connected that could definitely be a problem, but I would assume that they are connected through the stepper driver board at the very least.

I am considering refactoring the library code so it is more robust with respect to how the rx line is used. Perhaps there are cases where someone does not care about the responses and disconnects the rx line entirely. Or perhaps someone would like to use a better circuit that decouples the rx and tx line, which would make the library code cleaner since it would not have to remove the echoed data coming from the tx line.

musapinar, 2022-12-09:
I will try with a 10k resistor soon. Thank you @beamsjr 
I forgot to mention that it did not work any better with the SilentStepStick boards I received a few weeks ago @peterpolidoro So the problem must be me or my circuit (or both).

beamsjr, 2022-12-11:
@musapinar One other question, you have set up your v-ref correct? I had one that the VREF was not set and I thought it was, it was turned all the way to the right and it was reading some weird voltage but when I said it correctly it did start communicating.

musapinar, 2022-12-21:
Hello,
I figured while I was swapping my 1k resistor to a 10k one in my circuit, that the way I plugged the resistor was wrong. The schematic I drew above is good. Sorry for being a potatoe. I can confirm it works fine with either a 1k or a 10k resistor, but fails if you do not have any (or misplace it like I did...). I can confirm both TMC chips also work fine.
I regret wasting your time and I appreciate your effort troubleshooting this with me. Sorry.

peterpolidoro, 2022-12-21:
That is good news! I am glad it works for you now.

closed 2022-12-21 by peterpolidoro