janelia-arduino/TMC2209 #12: Unit of VELOCITY parameter of "stepper_driver.moveAtVelocity(VELOCITY);"

sseken, 2023-02-07:
Hello. I'm following your example of "TestMoveAtVelocity.ino". I added some lines in proper positions to communicate with TMC2209 as below. Now I can run and adjust the desired motor speed. 

const int32_t VELOCITY = 200;
const uint8_t RUN_CURRENT_PERCENT = 100;
const uint8_t MICROSTEPS_PER_STEP_EXPONENT = 0;

const TMC2209::SerialAddress SERIAL_ADDRESS_0 = TMC2209::SERIAL_ADDRESS_0;
stepper_driver.setup(serial_stream, SERIAL_BAUD_RATE, SERIAL_ADDRESS_0);
stepper_driver.setMicrostepsPerStepPowerOfTwo(MICROSTEPS_PER_STEP_EXPONENT);

My question is the unit of VELOCITY parameter of "stepper_driver.moveAtVelocity(VELOCITY);". I assumed that the unit of VELOCITY is "pulse per second", but it seems that there should be a factor of 1.4 to convert into pulse per second. Could you provide the detailed information about the parameter of "stepper_driver.moveAtVelocity()"?

peterpolidoro, 2023-02-07:
Hi. I added some documentation to the README to help explain moveAtVelocity.

The units of VELOCITY depend on the frequency of the clock used by the TMC2209. If the internal TMC2209 12 MHz clock is used, then you are right, microsteps_per_second = VELOCITY*0.715. Or as you noted, VELOCITY = microsteps_per_second*1.4.

I would like to perform that conversion automatically, but I do not know which clock frequency the user chooses.  Perhaps I can assume the default clock frequency and let the user change that value if necessary. Then I could use units of microsteps_per_second in the moveAtVelocity method, which would be more meaningful.

sseken, 2023-02-08:
@peterpolidoro Dear Peter Polidoro, Thank you so much for your kind help. I worried whether VELOCITY parameter is propotional to a real pulse per second. Now I feel free to use "moveAtVelocity()" function. Thank you once again.

closed 2023-02-08 by peterpolidoro

aagum-bae, 2024-05-16:
Hi, I know this is an old thread, however here is an excel file with the necessary calculations
[https://www.analog.com/media/en/engineering-tools/design-tools/TMC2209_Calculations.xlsx](url)

SoupySoups, 2024-06-21:
Gonna put into plain words for anyone who needs it
```cpp
  // steps_per_rev = 360/1.8 ( for a 1.8deg motor )
  // microsteps_per_rev = microsteps*steps_per_rev
  // velocity = microsteps_per_rev*1.4 (see https://github.com/janelia-arduino/TMC2209/issues/12)
  ```