janelia-arduino/TMC2209 #13: C porting

raulperez, 2023-02-15:
I have made a port in C for this library to make it work on STM32 devices in case somebody need it is available if the owner of the original library allows it.

I haven't published it on GitHub but if the original owner considers it is ok I will do it.

peterpolidoro, 2023-02-15:
Sure, all of my code is free open source, so you are welcome to publish your port.

I am planning on doing a significant rewrite in the near future. If you make your repository a fork of mine or somehow provide a reference then perhaps we each can continue to share improvements.

raulperez, 2023-02-15:
Sure, that would be great. I will gladly apply also the changes to the port. Or maybe I will do a port to C++ of the new code. I can also contribute if you want to the current implementation. It will help me a lot on my learning of the TMC2209.

closed 2023-02-15 by peterpolidoro