janelia-arduino/TMC2209 #14: INDEX output weird behaviour

raulperez, 2023-02-15:
I am using UART to communicate with my TMC2209 driver and motor is moving and all seems ok but I need to keep track of the steps through INDEX output. Is it normal that INDEX signal is getting active (HIGH) even it the motor is stopped?

I know that maybe this is not the right place to post my question, but since I am using this library maybe somebody has experienced the same behaviour as me.

I have these global_settings for my stepper:

[image]

raulperez, 2023-02-15:
I have managed for now to keep track of the count by using an external interrupt from on of the GPIOs when signal rise on INDEX and only taking in count when the motor is enabled and the direction of the motor. But still not clear why INDEX output is always oscillating between low and high even when the motor is not moving.

[image]

peterpolidoro, 2023-02-15:
I have never used the index pulse, so I am not sure how it is supposed to behave. How do you want it to behave? When you stop are you actively braking or passively braking or freewheeling? Maybe during active breaking the step generator is still active and sending pulses out over the index output.

raulperez, 2023-02-15:
In the TMC2209 datashee one can read this for index_step:

1: INDEX output shows step pulses from internal pulse generator. (toggle upon each step)

[image]

But instead it seems like it is a permanent signal. Does anybody knows what is happening?

raulperez, 2023-02-15:
I am not braking. Maybe you have a point. I am just using `disable` function.

I haven't configured the braking before. It is my first time working with stepper motors. Do you have any suggestions on how to setup the braking feature?

raulperez, 2023-02-15:
I found the issue. You were right 👍 I was stopping and running the motor with enable/disable function and I should instead use braking and VACTUAL set to 0 and INDEX started working with the expected behaviour.

Thanks you for the inspiration :) I have learnt a lot about TMC2209 today.

peterpolidoro, 2023-02-15:
Great! Glad to hear it. Good luck in your motor adventures.

closed 2023-02-15 by peterpolidoro