janelia-arduino/TMC2209 #15: Current Level Control when Using UART

ryancasler, 2023-02-26:
I'm having a bit of a problem when trying to use the UART control and it is around setting the current limit of the driver.  Whenever I use the UART interface, it stops using the on-board potentiometer setting for the current level control.  I have tried driving VREF with a current through the DAC on my board.  I've tried setting the internal sense registers and the external sense registers.  I've tried enabling analog current control and disabling it.  At this point, it doesn't matter which method I can get to work, I just want to set it up to be able to work.  Thanks!

peterpolidoro, 2023-02-27:
Can you try calling the method enableAnalogCurrentScaling() and see if the potentiometer works again?

mmin18, 2023-03-02:
Same issue here. I fixed it by adding stepper_driver.enableAutomaticCurrentScaling();
I don't know what this feature does, but seems it's been disabled when calling [setup()](https://github.com/janelia-arduino/TMC2209/blob/main/src/TMC2209.cpp#L30)

MikeGodinez, 2023-03-02:
I couldn't get my motor to move at all until I changed the pwm_autoscale and pwm_autograd bits in the PWMCONF register back to 1. Not sure if I also needed to set the pwm_autograd bit to 1, but that's the default at power-on.

I think it makes more sense to leave these on, since that's what they'll be set to if you use the TMC2209 without UART, so they're probably okay for most applications. Also, the note in section 6.2 of the datasheet (rev1.08) suggests only setting pwm_autoscale to 0 with a known motor and operating conditions, and you'll have to set PWM_GRAD and PWM_OFFSET to values that work.

peterpolidoro, 2023-03-03:
Good point about pwm_autoscale and pwm_autograd.

I originally kept them enabled by default, but there were a couple of occasions when those settings caused some motors to overheat. There seems to be certain types of motors where it is more difficult for the driver to calculate those values properly, but perhaps I need to read through the datasheet more carefully to figure out when and why. I disabled those settings by default so I would feel safe to attach any motor without worrying if would overheat, thinking I could always enable them later if necessary.

At the very least I need to add more documentation about this, thank you for bringing it to my attention.

peterpolidoro, 2023-03-03:
I updated the documentation to explain more about the settings and defaults. I am closing this issue for now, but please reopen if you run into more problems.

closed 2023-03-03 by peterpolidoro

ryancasler, 2023-04-25:
So, what exactly do I need to do to set the pwm_autoscale and pwm_autograd to 1?  I have still not gotten the motor to move at the current value set by the potentiometer.  When I don't use the UART connection it moves with about 800 mA of current but when I use UART it uses barely any and I can hold the stepper stationary with my hand without much effort.

peterpolidoro, 2023-04-25:
Have you tried using the methods:
enableAutomaticCurrentScaling()
enableAutomaticGradientAdaptation()

Can you please share the code you are using that does not seem to work?

Which stepper driver board are you using? If you are using a silent step stick, remember to solder the UART jumper pads on the back of the board to be able to use UART communication.