janelia-arduino/TMC2209 #16: Unable to Set Current Value via UART

sidn007, 2023-03-03:
The "CoolStep" example is perfectly working in my code. I have made zero changes to the said example code except keeping CoolStep  enabled throughout the execution of my code, but there's one kinda big hiccup. I'm unable to set the current. I went through the previously raised issues, but including enableAutomaticScaling or enableAnalogCurrentScaling made CoolStep NOT function as expected. Again, just running the "CoolStep" example as is, works PERFECTLY for me but I need higher current running through the motor. The current doesn't rise above 50mA when a load is applied to the motor. I just need a way to increase this current. HELP!

peterpolidoro, 2023-03-03:
So the CoolStep example alternates between enabling and disabling the cool step feature. How much current goes through your motor when cool step is disabled and how much does it drop when cool step is enabled? Does it have too much current going through your motor when cool step is disabled?

As far as I understand it, the entire point of the cool step feature is to drop the current significantly while the motor is spinning. If the current level is acceptable in your motor with cool step disabled then perhaps you should not use it at all in your application. If you do need cool step, though, maybe applying a load after the cool step is enabled confuses the driver. What happens if you disable cool step, get the motor up to speed, apply the load, and then enable cool step?

sidn007, 2023-03-03:
The load applied might be unexpected and may be applied at any random time. The idea is to keep CoolStep running all the time. When there is no load, the motor can operate at a lower current saving power and reducing the heat output. If and when a load is applied, CoolStep would increase the current momentarily till the load is removed. As I said, just increasing the current above 50mA would just work perfectly. Is there a way to increase the said 50mA current?

peterpolidoro, 2023-03-03:
Have you tried either enabling automatic current scaling and automatic gradient adaptation or adjusting the pwm offset and pwm gradient values?

sidn007, 2023-03-03:
Yes I did try that.  It kinda works... the motor starts spinning okay. But after a short while the current drops too low for some reason and the motor skips steps :(

peterpolidoro, 2023-03-03:
There are two options, I am not sure which one you tried.

One option is to leave both the automatic current scaling and automatic gradient scaling disabled, then slowly increase the pwm offset value to see if the current increases enough to handle the variable loads you are placing on it. If your motor is spinning very fast or has high back emf, then you may need to adjust the pwm gradient value as well. By adjusting both pwm offset and the pwm gradient values you should be able to get the current to increase.

The other option is to leave the pwm offset and pwm gradient values alone and instead enable automatic current scaling and automatic gradient adaptation. Then you can adjust the run current by using the setRunCurrent method.

Do either of those options work for you?

sidn007, 2023-03-03:
I tried enabling automatic current scaling and automatic gradient adaptation which resulted in the motor starting fine but then current drops and the motor starts to skip steps.

I'd like to try the pwm gradient and pwm offset. Since I'm new to this, I have no idea what the range of values are accepted. And it would be very helpful if you could explain what pwm gradient and pwm offset does.

Thanks!!

peterpolidoro, 2023-03-03:
I updated the documentation to explain more about the settings and defaults. I am closing this issue for now, but please reopen if you run into more problems.

sidn007, 2023-03-04:
Tried everything, current does increase when motor starts and then decreases after a while and motor starts skipping steps. Cool step isn't working at all when either pwm gradient and pwm offset or automatic current scaling and automatic gradient adaptation is enabled. The main intention of cool step is to increase current when load is applied. That's not working. As I said before, just letting cool step running on its own, works as intended (unfortunately only upto 50mA). I guess the default current value is 50mA. If there's a way to increase this "default" current value without tinkering with pwm gradient or pwm offset or automatic current scaling or automatic gradient adaptation, that would be the perfect solution. How can I do this?

MikeGodinez, 2023-03-04:
You tried raising COOLCONF.semin to prevent it skipping steps with automatic current scaling and automatic gradient adaptation? This is the first parameter to enableCoolStep(), and it's an integer from 1 to 15.
The CoolStep.ino example sets this very low, so, it runs as close to stalling the motor as possible.

sidn007, 2023-03-04:
> You tried raising COOLCONF.semin to prevent it skipping steps with automatic current scaling and automatic gradient adaptation turned on? This is the first parameter to enableCoolStep(), and it's an integer from 1 to 15. The CoolStep.ino example sets this very low, so, it runs as close to stalling the motor as possible.

If I'm not wrong, I should increase semin value in the TMC2209.h file right?... That too had no effect. The current still reduces close to 100mA while the motor is running.

sidn007, 2023-03-04:
okay, here's what I have set up right now. 

setRunCurrent(100)
setHoldCurrent(30)
setPwmOffset(255)
setPwmGradient(255)
disableAutomaticCurrentScaling()
disableAutomaticGradientAdaptation()
disableCoolStep()

So far this is what I've observed:
Stand Still current is working perfectly 
Current increases when motor starts and stops (which is exactly how I want it to work)
Current decreases when the motor runs at a medium to high speed
Current increases when the motor runs at a slow speed

Is there a way to invert this current assignment? (i.e., Current should increase as speed increases and current should decrease as speed decreases)
That would solve the whole issue :)

peterpolidoro, 2023-03-04:
What type of stepper motor are you using? Can you tell us the motor specs? What supply voltage are you using? How fast are you spinning the motor?

Your supply voltage may not be high enough to be able to source enough current to make up for the back-emf at speed.

sidn007, 2023-03-04:
I'm using the NEMA 17 stepper motor at 20V with a power supply capable of supplying constant 2A at ease. The motor spins at 1000 steps per second (Fullstep interpolated to 256). Previously I used the TMC2208 which I can confirm that it ran properly at about 1.1A, 20V

peterpolidoro, 2023-03-04:
What current values are you measuring through your motor with those settings? I have never used pwm offset and gradient values that are so large. Maybe it causes issues when they are totally maxed out. Can you start off with those values small and then slowly increase them and see how the motor current changes?

sidn007, 2023-03-04:
Tried changing the values in increments of 50. Still facing the same issue. Current drops to around 100mA when the motor is up to speed.

> What current values are you measuring through your motor with those settings? I have never used pwm offset and gradient values that are so large. Maybe it causes issues when they are totally maxed [...]

sidn007, 2023-03-04:
Tried changing the values in increments of 50. Still facing the same issue. Current drops to around 100mA when the motor is up to speed. Tried with 50, 100, 150, 200, 250, all had the same issue.

peterpolidoro, 2023-03-04:
Which TMC2209 board are you using? A silent step stick? Or did you make your own? Just curious what the values of the sense resistors are. Do you know what settings you used for the TMC2208 to get the current values you wanted?

sidn007, 2023-03-04:
I'm using the Bigtreetech TMC2209 v1.3. 
I used the TMC2208Stepper library by teemualut. Using rms_current( ), you can set the value of current that you exactly want. For example if I require 700mA of current, I'd just type driver.rms_current(700). Which would run the motor exactly at 700mA.

peterpolidoro, 2023-03-05:
I will look over the TMC2208 library when I get a chance and see if I can figure it out. In the mean time you may want to enable both the automatic current scaling and the automatic gradient adaptation and see if you can get it to work with the automatic tuning.

peterpolidoro, 2023-03-06:
I took a quick look at the TMC2208Stepper library. To match what that library does, try disableAnalogCurrentScaling(), enableAutomaticCurrentScaling(), enableAutomaticGradientAdaptation(), and then use setRunCurrent(percent) to see if you can increase the spinning motor current to whatever value you want. I do like that you can set the run current in milliamps in that library directly given the sense resistor values. I may add that calculation to this library when I get a chance.

sidn007, 2023-03-07:
On a quick observation on the A4988, TMC2208 and TMC220, here's what I've found out. The A4988 has the current constant irrespective of the motor state (standstill or running or even the motor speed). For some reason, the TMC2208 and the TMC2209 (both in standalone and UART mode) limits the current when the motor runs at a higher speed. Also it has a higher starting and stopping current when compared to the running current. The frustrating thing is that it doesn't allow the same higher current as the motor runs at a higher speed. For example, motor at 500 steps/sec has a higher current than motor at 1000 steps/sec. Not sure if this a hardware or a software limitation. It would be so great if the TMC drivers provide constant current as the A4988!

peterpolidoro, 2023-03-07:
So did you find this out by testing all three drivers with the same motor? Are you applying loads to the motors when they are at speed or just running them at various velocities with no load?

It is unclear to me why you would want higher current through the motors when you are spinning them at high speeds. Ideally the driver should use just enough current to keep the motor spinning at the commanded speed and only increasing the current to compensate for increased load so that the velocity stays correct. Normally you would want to minimize current to minimize heating and electrical losses. Using higher current for the same load and same velocity seems like a bad thing most of the time.

You could contact the Trinamic tech support, though, and ask them about their drivers and why they behave differently than the A4988. If you find out anything from them or figure out anything that needs to be changed in this library to get it to work like you want then please let me know.

peterpolidoro, 2023-04-28:
I have just released a new version 9.0.0. When you have a chance, please test the new changes and see if the new version resolves any issues you may be having with the previous version. If you find any bugs or would like to request any new features please leave me a comment or open a new issue. Thank you!

closed 2023-06-27 by peterpolidoro