janelia-arduino/TMC2209 #17: Might have a bug while testing/

Zhehua-Hu, 2023-03-05:
Repeat kernal Code：
AccelStepper Xaxis(1, 16, 17); // step, direction
AccelStepper Yaxis(1, 4, 26); // step, direction
int x_circle = 1600；

in setup
    Xaxis.moveTo(2*x_circle);
    Yaxis.moveTo(-2*x_circle);

in loop:
      Xaxis.runToPosition();
     Yaxis.runToPosition();

While X move(negtive) && Y move(negtive) ,the motor will not run.
By my test, the issue will happen when both of two motor move is negtive. When this is one negtive,work well.

Zhehua-Hu, 2023-03-05:
@peterpolidoro May you help to debug?

peterpolidoro, 2023-03-06:
I have never used the AccelStepper library before so I am not sure I have any advice to give you about that. Are you saying you think there is a bug in the AccelStepper library or are you saying you think there is a bug in the TMC2209 library?

peterpolidoro, 2023-04-28:
I have just released a new version 9.0.0. When you have a chance, please test the new changes and see if the new version resolves any issues you may be having with the previous version. If you find any bugs or would like to request any new features please leave me a comment or open a new issue. Thank you!

closed 2023-06-27 by peterpolidoro