janelia-arduino/TMC2209 #18: Stepper changes speed or resolution alone

4mInnovations, 2023-03-20:
Hi,
I'm trying to use a tmc2209 drive in uart mode with an arduino Pro Mini (with AltSerial).

It works well for one of my 2 motors, but for the other one, wich is running a paper winder, the stepper works fine for a few cycles, then after that the stepper speed drops very slow on its own.

I use the library accelStepper for the both steppers.
Also, I have an encoder that works with the interrupts.

Do you have any idea what could be causing this problem?

peterpolidoro, 2023-03-20:
That is odd behavior. Is it possible for you to disconnect the motor from the paper winder and see if it spins properly with no load on it? Or do you have another identical stepper that you can try in its place? Does the load on the paper winder stay constant? Maybe the driver is overheating and shutting down. Or perhaps something in the firmware is causing a problem, but I have not used accelStepper much so I do not know what could cause it to malfunction. Perhaps try disconnecting the encoder, maybe the interrupts occur so frequently that they mess up the step generation. In general, you may want to try testing each piece of the system as isolated as you can make it. Try disconnecting the second stepper and the encoder and just try spinning the paper winder on its own and see how that behaves, that sort of thing. Let me know how it goes and I will try thinking of more things to test.

4mInnovations, 2023-03-20:
Hi,
1) yes the motor spins properly with no load on it
2) yes I have another motor, I will try it
3) I don't think the load on the paper stay constant
4) I will try the stepper alone, i.e. no other stepper or encoder in the arduino program

I'm come back with the results.

peterpolidoro, 2023-03-20:
If the motor spins properly with no load and the load on the paper winder is not constant, then you may need to play with the driver settings in order for it to work properly for all of the loads on the motor.

The driver can operate in either voltage control mode (automatic current scaling disabled) or current control mode (automatic current scaling enabled). For variable loads, you probably want to be using current control mode. The default of this library is voltage control mode. If you have not already, also try enabling automatic current scaling. There is documentation about it in the driver datasheet and also on the README of this repository.

peterpolidoro, 2023-03-20:
> Ok,
> I would try the current control mode, but I try to put in my arduino code:
> driverTMC.enableAutomaticCurrentScaling();
> And the ide said:
> Compilation error: 'class TMC2209Stepper' has no member named 'enableAutomaticCurrentScaling'

Do you know which version of the library you are using? The latest version is 8.0.7. You may need to upgrade to the newest version.

peterpolidoro, 2023-03-20:
It is on my todo list, but right now it only works with hardware serial, sorry! I hope to change that in the next few weeks.

4mInnovations, 2023-03-20:
It's ok, thank you so much for your answers.
I will probably test my setup with an arduino mega (with 3 hardware serials) and when you will add the soft serial, I will re-change for my Pro Mini (I don't have enough space for use a mega in my final setup)

peterpolidoro, 2023-03-20:
Great, please let me know how it goes. There are other Arduino-compatible small boards with multiple hardware serials, Teensy 4 for example.

4mInnovations, 2023-03-21:
Hi,
I try the library TMC2209 with an arduino Mega and two steppers motors.
I can setup the 2 drives and I can communicate with them :)

But, I don't understand how to manage the enable (enable pin), because I use accelstepper that have already enable:
stepper0.setPinsInverted(false, false, true);    
stepper0.enableOutputs();

I find you have also a enable() function and a disabledByInputPin() in your TMC2209 library.

peterpolidoro, 2023-03-22:
The TMC2209 can be disabled in two ways, hardware and software.

You need to enable the TMC2209 in both hardware, bringing the enable pin high, and in software, by calling the enable() method.

I added some documentation about both types of enabling in the README.

4mInnovations, 2023-03-22:
Ok,
but I read the readme file and I don't find anything about enable.

In the TMC2209.h, you have:
  // driver must be enabled before use it is disabled by default
  void enable();
  void disable();

  // driver may also be disabled by the hardware enable input pin
  // this pin must be grounded or disconnected before driver may be enabled
  bool disabledByInputPin();

peterpolidoro, 2023-03-22:
Ah yes sorry, I updated the README, but forgot to push the changes to the repository properly. It should be fixed now.

Basically you need to enable the driver in both software and hardware. Enable the driver in software by calling enable() and also pull the enable line high if it is not already.

4mInnovations, 2023-03-23:
Hello Peter,
I finally managed to get 2 drives and 2 steppers to work with the TMC2209 library (UART and dir/step pins) and a arduino Mega. 

I will wait to be able to use the altSoftSerial library (or other) with the library TMC2209 and my arduino Pro Mini.

I have two others questions:
   1) I need to run my stepper at higher amperage because right now, my stepper can't perform the mechanical work. My oldest setup with DRV8825, I was using higher values of current to perfom the work.

   2) In another project, I use an adafruit Grand central M4 (SAMD51) for control 4 steppers. I have setup my grand central to have Serial2 and Serial3 functional. I try the same setup as with my Mega.  I can communicate with the drive at the address 0, but I can't communicate with the drive at the address 1. I know for sure that Mega is on 5V and the grand central is on the 3.3V. Do you have any idea why this is not working ? (the resistor of 1k ??)

peterpolidoro, 2023-03-23:
If you are using the driver in current control mode, then you should be able to increase the current with the setRunCurrent(percent) method. Try increasing the run current percent value and see what happens.

For your other project, are you saying that you have four stepper drivers, each connected to its own serial port? Or are you connecting multiple drivers to a single serial port? If each driver has its own port, then you can use address 0 for each driver. If you are connecting multiple drivers to a single serial port, then you need to set a different address for each driver by setting the address pins high or low on each driver. The 1k resistor may not be the best value for all microcontrollers, but I have gotten it to work on a Teensy 4.0 that runs at 3.3V. You can try changing the resistor value, but it sounds like you may have another issue in your setup.

4mInnovations, 2023-03-23:
For the current, I use in current control mode and the setRunCurrent(percent) is already at 100%:
const uint8_t RUN_CURRENT_PERCENT = 100;
const uint8_t HOLD_CURRENT_PERCENT = 100;
const uint8_t HOLD_CURRENT_PERCENT_REDUCED = 10;    
    
    driverPapier.enableAutomaticCurrentScaling();
    driverPapier.enableAutomaticGradientAdaptation();
    driverPapier.setRunCurrent(RUN_CURRENT_PERCENT);
    driverPapier.setHoldCurrent(HOLD_CURRENT_PERCENT);

For my other project, I tried with the same setup that it works with the mega, i.e. 2 drives in the same serial port.

TMC2209 stepper_driver_0;
const TMC2209::SerialAddress SERIAL_ADDRESS_0 = TMC2209::SERIAL_ADDRESS_0;
TMC2209 stepper_driver_1;
const TMC2209::SerialAddress SERIAL_ADDRESS_1 = TMC2209::SERIAL_ADDRESS_1;
const long SERIALX_BAUD_RATE = 115200;

and in the setup:
stepper_driver_0.setup(Serial2,SERIALX_BAUD_RATE,SERIAL_ADDRESS_0);
stepper_driver_1.setup(Serial2,SERIALX_BAUD_RATE,SERIAL_ADDRESS_1);

4mInnovations, 2023-03-24:
Hi, I don't understand at all.
My setup with 2 drives, 2 motors, Serial2 selection work fine with my arduino mega 5V.

I use exactly the same setup, but I use a adafruit grand central M4 (at 3.3V). I can communicate with the driver at the address 0, but not the driver at the address 1.
-I try to change the resistor of 1k to 470 ohms
-I try to change the baud rate

peterpolidoro, 2023-03-30:
> Hi Peter,
> I recheck all my setup:
>
> With my breadboard, I wire my arduino Mega and I try my software. My 2 drives setup and communicate.
>
> With the same breadboard (not changing anything), I wire my adafruit grand central M4 and I try the same software. My drive at the address 0 setup and communicate. My drive at the address 1 not setup and not communicate.
>
> I try many baud rate, all the same.
> Do you have any idea?
> I try also to setup/communicate one drive at the time at the address 0 (switch the MS1 and MS2 between the setup).
> If I launch the setup of the drive0 in first, the drive 0 work and the drive1 not work.
> If I launch the setup of the drive1 in first, the drive 1 work and the drive0 not work.

What TMC2209 driver board are you using?

4mInnovations, 2023-03-30:
I try the BigTreetech TMC2209 v1.2

peterpolidoro, 2023-04-07:
Maybe this is an issue with the SENDDELAY value (the delay until a reply is sent). 

From the datasheet:
> To ensure a clean bus transition from the master to the node, the TMC2209 does not immediately send
the reply to a read access, but it uses a programmable delay time after which the first reply byte
becomes sent following a read request. This delay time can be set in multiples of eight bit times using
SENDDELAY time setting (default=8 bit times) according to the needs of the master. In a multi-node
system, set SENDDELAY to min. 2 for all nodes. Otherwise, a non-addressed node might detect a
transmission error upon read access to a different node.

I have never tried changing this value and I would need to make some changes to this library in order to do so.

You could try adding hardware switches  to each device like it suggests in the datasheet, that way you can just use a single UART, but then use port pins to select which device is active, similar to SPI.

peterpolidoro, 2023-04-28:
I have just released a new version 9.0.0. When you have a chance, please test the new changes and see if the new version resolves any issues you may be having with the previous version. If you find any bugs or would like to request any new features please leave me a comment or open a new issue. Thank you!

4mInnovations, 2023-05-01:
Hi Peter,
What do you change?
I will try today or tomorrow, but I want to know if you change something for the virtuel port (pro mini arduino) or/and the serial comm (grand central m4 arduino)?

Envoyé à partir de Outlook pour iOS<https://aka.ms/o0ukef>

4mInnovations, 2023-05-02:
Hi Peter,
I try this morning your new library, with:
- 2x TMC2209
- 2x Stepper 
- Arduino 5V pro Mini
- SoftwareSerial on pin 8 and 9
- 2x address differents for driver (00 and 01)
- One way communication (no Rx)

All seems working, I change the resolution of each motor and I can see the difference.
Thanks a lot for this.

I will try the same setup, but with an arduino grand central M4 3.3V probably tomorrow.

4mInnovations, 2023-05-05:
It's not good, I try  this morning and I have a compilation error from the SoftwareSerial.h for the Grand central (SAMD51)

In file included from C:\Arduino\test library TMC2209\test_TMC2209Lib_v7a_GCM4\test_TMC2209Lib_v7a_GCM4.ino:8:
c:\Arduino\libraries\TMC2209\src/TMC2209.h:11:10: fatal error: SoftwareSerial.h: No such file or directory
   11 | #include <SoftwareSerial.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.

peterpolidoro, 2023-05-05:
I was under the impression that SoftwareSerial does not work on ESP32 or SAMD_SERIES boards, so I excluded those from using SoftwareSerial with a preprocessor directive. You have used SoftwareSerial with a SAMD board and it works? If you make an example ino file and just include SoftwareSerial and do not include TMC2209 does it compile?

4mInnovations, 2023-05-05:
You are right, SoftwareSerial not work on SAMD series boards. 

I don't understand, I not use SoftwareSerial in my code, the first one try I've got the error.
I restart the ide, update the library and all seem to work (it's compile). :)

I have another thing to do today, I will try it monday. 
Thanks again.

closed 2023-06-27 by peterpolidoro