janelia-arduino/TMC2209 #19: Feature Request: Alternative RX/TX Pins for UART

LenStruttmann, 2023-04-26:
So far, I'm liking your lib. However, it appears that you require the Serial connection to the TMC2209 to use the standard RX/TX pins.  To get your lib to work on an Heltec ESP32 Dev Kit V3, I needed to alter your code to allow me to redefine the pin used for RX.

The Heltec dev boards only provide access to Serial2 and they use the standard U2_RXD pin for the onboard OLED display.  To use Serial2, one must define the RX pin to use one of the other unused GPIO pins.

peterpolidoro, 2023-04-26:
Hi,
So I will be releasing a new version of this library in the next day or two. The receive line will be optional for all unidirectional commands that do not care about a response from the TMC2209.

The current version of this library does not make any assumptions about which RX and TX lines are used, however. This version does require that hardware serial is used. The next version will allow you to use software serial for unidirectional commands. Some microcontrollers allow you to remap functionality from one pin to another. Maybe it is possible for you to remap the hardware serial RX pin on your microcontroller to another pin which will allow it to work with the current version of this library.

LenStruttmann, 2023-04-27:
Thanks for the reply!

_The_ current version of this library does not make any assumptions about which RX and TX lines are used, however.

Well, yes it does.  It assumes that the user is using the default RX and TX pins for the selected Hardware Serial.

 _Maybe it is possible for you to remap the hardware serial RX pin on your microcontroller to another pin which will allow it to work with the current version of this library._

Yes, that is what I do, BUT, the only way to do that is to provide the RX and TX pins via the call:

serial_ptr_->begin()

in:

void TMC2209::setOperationModeToSerial()

Since this function is private, there is no way for the user to specify those pins.

To get you library to work, I updated:

TMC2209::setOperationModeToSerial()
and
void TMC2209::setup()

... to allow me to pass the alternative RX and TX pins.

Just something for you to think about to allow ESP32 users to be able to easily use you rlib.

Thanks!

peterpolidoro, 2023-04-27:
Oh I see, yes that makes sense.

If you would like to share your modifications I can add them to the next version of the library for you.

LenStruttmann, 2023-04-27:
Here are the changes that I made to allow your lib to work on my development setup (PlatformIO, Heltec ESP32 Dev Kit V3):

**File: TMC2209.h**

  void setup(HardwareSerial & serial,
    long serial_baud_rate=115200,
    SerialAddress serial_address=SERIAL_ADDRESS_0, int8_t rxPin=-1, int8_t txPin=-1);

...and...

 void setOperationModeToSerial(HardwareSerial & serial,
    long serial_baud_rate,
    SerialAddress serial_address=SERIAL_ADDRESS_0, int8_t rxPin=-1, int8_t txPin=-1);

**File: TMC2209.cpp**
void TMC2209::setup(HardwareSerial & serial,
  long serial_baud_rate,
  SerialAddress serial_address, int8_t rxPin, int8_t txPin)
{
  blocking_ = false;
  serial_baud_rate_ = serial_baud_rate;
  setOperationModeToSerial(serial, serial_baud_rate, serial_address, rxPin, txPin);

...and...

void TMC2209::setOperationModeToSerial(HardwareSerial & serial,
  long serial_baud_rate,
  SerialAddress serial_address, int8_t rxPin, int8_t txPin)
{
  serial_ptr_ = &serial;
  serial_address_ = serial_address;

  serial_ptr_->begin(serial_baud_rate, SERIAL_8N1, rxPin, txPin );

**Here is the setup call in my code:**
#define RX_PIN  5
#define TX_PIN 26
  stepper_driver.setup(serial_stream, 115200, TMC2209::SERIAL_ADDRESS_0, RX_PIN, TX_PIN );

I am a C++ newbie, so I do not know how to avoid the requirement to specify "115200, TMC2209::SERIAL_ADDRESS_0, " in that above call without re-ordering the arguments.

Thanks!

peterpolidoro, 2023-04-28:
I have just released a new version 9.0.0. When you have a chance, please test the new changes and see if the new version resolves any issues you may be having with the previous version. If you find any bugs or would like to request any new features please leave me a comment or open a new issue. Thank you!

LenStruttmann, 2023-05-02:
Unfortunately, the "#include SoftwareSerial.h" in V9.0.0 does not seem to be compatible with ESP32 MCUs.  I'm about to leave for a business trip, so I do not have time right now to investigate further.  See: https://stackoverflow.com/questions/60094545/esp32-softwareserial-library

closed 2023-06-27 by peterpolidoro