janelia-arduino/TMC2209 #26: Compilation error: 'Serial1' was not declared in this scope

qq331212627, 2023-07-11:
I use arduino uno+tmc2209, uart mode.

Each example reports the same error during compilation
------------------------------------------------------------------------------------------------------------
C:\Users\Administrator\AppData\Local\Temp\.arduinoIDE-unsaved2023611-17332-1axbd5w.itbwh\TestCommunication\TestCommunication.ino:3:34: error: 'Serial1' was not declared in this scope
 HardwareSerial & serial_stream = Serial1;
                                  ^~~~~~~
C:\Users\Administrator\AppData\Local\Temp\.arduinoIDE-unsaved2023611-17332-1axbd5w.itbwh\TestCommunication\TestCommunication.ino:3:34: note: suggested alternative: 'Serial'
 HardwareSerial & serial_stream = Serial1;
                                  ^~~~~~~
                                  Serial

exit status 1
--------------------------------------------------------------------------------------------------------------
Compilation error: 'Serial1' was not declared in this scope

peterpolidoro, 2023-07-11:
Yes the Uno does not have a Serial1 port: [Arduino Serial](https://www.arduino.cc/reference/en/language/functions/communication/serial/)

> On Uno, Nano, Mini, and Mega, pins 0 and 1 are used for communication with the computer. Connecting anything to these pins can interfere with that communication, including causing failed uploads to the board.

To use the Uno with this library, you must use SoftwareSerial. SoftwareSerial is more limited than HardwareSerial, but it allows you to set parameters on the TMC2209. SoftwareSerial cannot get information back from the TMC2209, only unidirectional communication is possible.

If you need bidirectional communication with the TMC2209, you should pick another Arduino board with an additional HardwareSerial interface.

closed 2023-07-11 by peterpolidoro

qq331212627, 2023-07-12:
Thank you for your reply

It's very detailed

I've got it moving in uart mode