janelia-arduino/TMC2209 #27: PNG images in the repository are hard to read with a dark theme

berkut0, 2023-07-25:
Speaks for itself:

[image]

peterpolidoro, 2023-07-31:
Oh great point. I had some trouble in the past with svg images not displaying properly, but I was not thinking about dark themes. I could either convert all images to svg, change the black lines in the images to some other color, or stop using transparent background in the png images so they will look like big white blocks on a dark theme background. Which option do you think would be best?

berkut0, 2023-08-02:
Interesting question, but to be honest I don't know. Perhaps the best solution will be to choose colour for dark elements they would be visible. But I think it is an overcomplicated solution, and just making them backgrounds would be simple and robust. Would be great to see another opinion.

peterpolidoro, 2023-08-02:
I redid the images, saving them as svg files rather than png files. I modified the color scheme, so hopefully the lines and text are now visible when using either a light or dark theme. Check them out and see what you think. Feel free to open the issue again if you still have trouble reading them or if you find any mistakes. Thanks!

closed 2023-08-02 by peterpolidoro