janelia-arduino/TMC2209 #28: Is driver current possible to set explicitly?

berkut0, 2023-07-25:
Not sure, but apparently there is no way to set the current explicitly, right?
It seems like there is only percentage and scaling

peterpolidoro, 2023-07-31:
In order to set the current explicitly, the function must know the value of the current sense resistors. Many TMC2209 driver boards use typical values for the current sense resistors and I could use those values as defaults, but there would need to be a way to set different current sense resistor values as well. I could add something like that if you would find it useful.

ShadeTechnik, 2023-08-02:
I was also curious about this. This is my first project using TMC drivers but I noticed Marlin has a parameter for setting the value of the external resistor. I use the BTT EZ2209 driver and I'm enabling the external resistors by disabling the internal resistors......but I'm not sure how to configure the resistor value.

berkut0, 2023-08-02:
As I understand it, there is something called a scaler inside the driver? And essentially we're putting out a current multiplier? The driver itself doesn't know the current, but it can detect it and provide a proportion of it with a scaler? I tried the more popular library first, it has current. But after studying the source code I realised that the current is just calculated from the scaler values and a predefined Rsense. Right?

berkut0, 2023-08-02:
Maybe the percentages are a bit confusing because there is not as much control in fact

[image]

peterpolidoro, 2023-08-04:
I do agree that it would be useful to be able to set the current directly, but I will have to dig into the datasheet to make sure that is being calculated properly in all cases.

In current control mode, you would need to know the value of the sense resistor and perhaps also the voltage of the power supply. It would only work properly if the automatic calibration performed by the driver worked well. I have found that for some motors this calibration does not always work well, but the datasheet describes steps to help ensure it happens properly.

In voltage control mode, you would need to know the voltage of the power supply and the properties of the motor to convert the output voltage into motor current.

It might get too complicated to do well, but I will look into it.

ShadeTechnik, 2023-08-05:
I have a hard time following the TMC datasheet, it gets complicated when you dig in. I don't know if specifying the sense resistor value is necessary, I just know that it is done in some 3D printer firmwares likely for ease of use [https://marlinfw.org/docs/hardware/tmc_drivers.html](url)

By "voltage control mode" I assume you are referring to "velocity based scaling". I think that's beyond the scope, it doesn't have anything to do with Rsense resistors. Rsense is based on current feedback compared to VREF. Either the driver chip is using internal resistors that don't account for the driver being part of a module, or the module manufacturer has selected Rsense resistors to limit the chip's maximum output to what the driver module can handle. The little step stick type modules usually use 0.11r resistors to limit the max output for something reasonable for a tiny PCB

I don't fully understand the TMC driver settings yet so maybe some automatic whatever accounts for the Rsense value

s-light, 2025-06-23:
is this issue solved with this function?!
https://github.com/janelia-arduino/TMC2209/blob/main/src/TMC2209/TMC2209.cpp#L239

and we only need some documentation for this?