janelia-arduino/TMC2209 #32: Timing issue setting multiple drives

ShadeTechnik, 2023-08-05:
I have to preface this by saying "I'm not a code guy" myself, a friend is helping with this....

I designed a PCB using upto 6 BTT EZ2209 drivers (hardware can address 8). I did this using the suggestion in the TMC manual on page 21 using the analog switches and the buffer. The board has an Attiny 3226 that's only job is to program the 2209's on startup and eventually take in parameter changes over UART or SPI. ATM the code only sets up the drives.

When testing I could not get the timing right at all using the buffer IC so I removed it, now it is only the analog switch ICs selecting each drive prior to programming. This worked but the first drive would often report no settings and most of the time for whatever reason none of the microstep settings ever took. There were delay()s in the code to allow things to settle before programming the drives. Removing all of the delays helped some and I have to keep the replyDelay setting @ 2

with the current code, the drives will generally all program successfully except for the microsteps_per_step setting, for some reason this is 50/50 whether it takes or not. is_setup also comes back 0 about half the time, otherwise they all set. I'm trying to figure out if the library is just not friendly for this use or I have some timing issue yet. The code is here:
[https://github.com/ShadeTechnik/ST-7i92-PnP-BoB/blob/main/step-drive-mcu/src/step-drive/step-drive.ino](url)

ShadeTechnik, 2023-08-05:
An example print from the debug port when microsteps fail to set:
`Debug Console Running....
Setting Drive 0 Now......
Retrieving Drive 0 Settings......
settings.is_setup = 0
settings.irun_percent = 48
settings.ihold_percent = 29
settings.microsteps_per_step = 8
settings.cool_step_enabled = 1
stall_guard_threshold = 50
settings.internal_sense_resistors_enabled = 0
settings.stealth_chop_enabled = 1
settings.automatic_current_scaling_enabled = 1
*************************

Setting Drive 1 Now......
Retrieving Drive 1 Settings......
settings.is_setup = 0
settings.irun_percent = 100
settings.ihold_percent = 48
settings.microsteps_per_step = 8
settings.cool_step_enabled = 1
stall_guard_threshold = 50
settings.internal_sense_resistors_enabled = 0
settings.stealth_chop_enabled = 1
settings.automatic_current_scaling_enabled = 1
*************************

Setting Drive 2 Now......
Retrieving Drive 2 Settings......
settings.is_setup = 0
settings.irun_percent = 48
settings.ihold_percent = 29
settings.microsteps_per_step = 8
settings.cool_step_enabled = 1
stall_guard_threshold = 50
settings.internal_sense_resistors_enabled = 0
settings.stealth_chop_enabled = 1
settings.automatic_current_scaling_enabled = 1
*************************

Setting Drive 3 Now......
Retrieving Drive 3 Settings......
settings.is_setup = 0
settings.irun_percent = 19
settings.ihold_percent = 19
settings.microsteps_per_step = 8
settings.cool_step_enabled = 1
stall_guard_threshold = 50
settings.internal_sense_resistors_enabled = 0
settings.stealth_chop_enabled = 1
settings.automatic_current_scaling_enabled = 1
*************************

Setting Drive 4 Now......
Retrieving Drive 4 Settings......
settings.is_setup = 0
settings.irun_percent = 19
settings.ihold_percent = 19
settings.microsteps_per_step = 8
settings.cool_step_enabled = 1
stall_guard_threshold = 50
settings.internal_sense_resistors_enabled = 0
settings.stealth_chop_enabled = 1
settings.automatic_current_scaling_enabled = 1
*************************`

peterpolidoro, 2023-08-05:
Hi, have you tried changing the reply delay value for each chip? Look in the README for:

- [When multiple TMC2209 chips are connected to the same serial line with multiple addresses then the reply delay value should be increased, otherwise a non-addressed chip might detect a transmission error upon read access to a different chip.]

ShadeTechnik, 2023-08-05:
>
> 
> .........same serial line with multiple addresses

.......is a different setup. 4 TMC2209's can be addressed from a single line, I'm not doing that. I'm "ADDRESSING MULTIPLE NODES" with a 74HC1G125 as shown on  P21. Basically the UART is running through a quad switch IC and an IO pin controls each switch. Connect to one drive, send config, read config, disconnect.....then repeat. No reason for different reply delays as the line is only open to 1 drive at a time.

I think I resolved this in my code with a delay() before sendDriveConfig. I'll give it a day or so then I'll close the issue.

peterpolidoro, 2023-08-05:
Can you show me in the code where the delay seems to be needed? Thanks.

ShadeTechnik, 2023-08-05:
Right above send_drive_config on line 193 [https://github.com/ShadeTechnik/ST-7i92-PnP-BoB/blob/2607dd1fa31b6f57d524a89980e9e26ed283a25a/step-drive-mcu/src/step-drive/step-drive.ino#L193](url)

I'll update that code soon as I get a chance