janelia-arduino/TMC2209 #37: About the “sense resister” value, namely “`R_SENSE`.”

issuefiler, 2023-09-05:
```cpp
inline constexpr float sense_resister_value = 0.11F;
inline constexpr std::uint8_t uart_slave_address = 0b00;

TMC2209Stepper driver {&Serial1, sense_resister_value, uart_slave_address};
```

 When I used the _teemuatlut/TMCStepper_ library, I needed a value called `R_SENSE` to construct a `TMC2209Stepper` instance. I set it with the common value `0.11F` without actually understanding what that means and what that does.

 Using this library, _janelia-arduino/TMC2209_, I’ve noticed that the `TMC2209` constructor doesn’t require the `R_SENSE` value, unlike the `TMC2209Stepper`.

 I wonder why. How is the value used in the _teemuatlut/TMCStepper_ library and why is it not used in the _janelia-arduino/TMC2209_ library? Is it unnecessary? What does that value mean and do?

peterpolidoro, 2023-09-21:
The sense resistor value is probably an input to that library so that motor current can be set using units of amps or milliamps rather than simply a register value or percentage of maximum current.

I have been thinking of implementing a similar function in this library, but I need to understand all of the details of how that would work exactly before adding it.

issuefiler, 2023-09-24:
Yes, so I noticed. I’ve been reading their code and Trinamic’s datasheet thoroughly and I’ve seen their `rms_current` method calculating the most efficient `IRUN`, `IHOLD`, and `vsense` values out of the `R_SENSE` value and the given amperage.

 Closing this issue, as it’s a duplicate of the issue #28.

closed 2023-09-24 by issuefiler