janelia-arduino/TMC2209 #38: Bi-directional communication issue / RP2040 stallts / does not boot

nmaas87, 2023-10-22:
Good day,
I setup an RP2040 with 2x TMC2209 v1.3 (Bigtree Tech) drivers exactly to ( https://github.com/janelia-arduino/TMC2209#connecting-multiple-tmc2209-chips / - Bidirectional Communication / Chips needing different settings using one UART ).

I can confirm when I run following script and only setup one of the drivers and run their loop routines, it will work - does not matter which of both chips I choose, its ok. However, as soon as I enable both setups and the loop routines as shown below, the RP2040 will not boot at all. Windows Device Manager will show it to be a unknown, erronous USB device. To recover, I need to unplug, hold BOOTSEL and reflash via UF2 or use the nuke.uf2 to reset the chip.

Do you have an idea what I am doing wrong? Thanks a lot :)

````

#include <TMC2209.h>

// This example will not work on Arduino boards without HardwareSerial ports,
// such as the Uno, Nano, and Mini.
//
// See this reference for more details:
// https://www.arduino.cc/reference/en/language/functions/communication/serial/

//SerialPIO(txpin, rxpin, fifosize (32 bytes default)) 
SerialPIO Serial3(2,3);

//HardwareSerial & serial_stream = Serial3;
const int DELAY = 500;

// Instantiate the two TMC2209
TMC2209 stepper_driver_0;
const TMC2209::SerialAddress SERIAL_ADDRESS_0 = TMC2209::SERIAL_ADDRESS_0;
TMC2209 stepper_driver_1;
const TMC2209::SerialAddress SERIAL_ADDRESS_1 = TMC2209::SERIAL_ADDRESS_1;
const uint8_t REPLY_DELAY = 4;
const long SERIAL_BAUD_RATE = 115200;

void setup()
{
  Serial.begin(SERIAL_BAUD_RATE);
  Serial.ignoreFlowControl(true);
  // TMC2209::SERIAL_ADDRESS_0 is used by default if not specified
  stepper_driver_0.setup(Serial3,SERIAL_BAUD_RATE,SERIAL_ADDRESS_0);
  stepper_driver_0.setReplyDelay(REPLY_DELAY);
  stepper_driver_1.setup(Serial3,SERIAL_BAUD_RATE,SERIAL_ADDRESS_1);
  stepper_driver_1.setReplyDelay(REPLY_DELAY);  
}

void loop()
{

  if (stepper_driver_0.isSetupAndCommunicating())
  {
    Serial.println("Stepper driver is setup and communicating!");
    Serial.println("Try turning driver power off to see what happens.");
  }
  else if (stepper_driver_0.isCommunicatingButNotSetup())
  {
    Serial.println("Stepper driver is communicating but not setup!");
    Serial.println("Running setup again...");
    //stepper_driver.setup(serial_stream);
    stepper_driver_0.setup(Serial3,SERIAL_BAUD_RATE,SERIAL_ADDRESS_1);
    //stepper_driver_0.setReplyDelay(REPLY_DELAY);  
  }
  else
  {
    Serial.println("Stepper driver is not communicating!");
    Serial.println("Try turning driver power on to see what happens.");
  }
  Serial.println();

  
  if (stepper_driver_1.isSetupAndCommunicating())
  {
    Serial.println("Stepper driver is setup and communicating!");
    Serial.println("Try turning driver power off to see what happens.");
  }
  else if (stepper_driver_1.isCommunicatingButNotSetup())
  {
    Serial.println("Stepper driver is communicating but not setup!");
    Serial.println("Running setup again...");
    //stepper_driver.setup(serial_stream);
    stepper_driver_1.setup(Serial3,SERIAL_BAUD_RATE,SERIAL_ADDRESS_1);
    //stepper_driver_1.setReplyDelay(REPLY_DELAY);  
  }
  else
  {
    Serial.println("Stepper driver is not communicating!");
    Serial.println("Try turning driver power on to see what happens.");
  }
  Serial.println();

  
  delay(DELAY);
}

````

nmaas87, 2023-10-22:
Some additional infos: Running Arduino with latest arduino-pico 3.6.0 version.

I also tested to have multiple chips attached to multiple UART interfaces, this does work without issue, it really looks that multiple drivers on one line with different addresses seems to provoke some issue.

Another thing I realizied using the example BidrectionalCommunication -> TestCommunication: Not matter the setup, if you're booting the RP2040 with the power supply to the TMC2209 switched off, the example will correctly state 

````
Stepper driver is not communicating!
Try turning driver power on to see what happens.
````

If you then turn on the power, following message appears:

````
Stepper driver is communicating but not setup!
Running setup again...
````

At this point the RP2040 stalls, only a reboot helps to get any communication back. Also the serial terminal in the Arduino software locks up, meaning you cannot even close the window. Only after several tries this will work (if it all) - or only disconnecting the RP2040 will help at all.

Thank you :)

peterpolidoro, 2023-10-23:
Hi. Just to be clear, do you have the address line on the second driver pulled high to set the proper address when you gave two drivers connected to the same UART?

nmaas87, 2023-10-23:
That is correct. I also could add my driver at addr 0, access it alone,
just use the addr 1 driver alone and this also worked, but as soon as I
have both drivers on the same UART at the same time (one with 0 one with 1)
it does not work anymore.

nmaas87, 2023-10-23:
I tried this and this seems to move into the correct direction! 👍🏻 

One thing to note, it does not like at all having run the setReplyDelay(REPLY_DELAY); during the loop again, I guess its enough to run it on setup and then it should be ok? Because otherwise the board communication becomes quite unstable.

Also, addresses 0 and 1 answer now, I see no stalling board anymore, so this is great - however, I can see that after having the drivers successfully configured for the first time and shutting the drivers 12v off and on again, the board tends to "swing", really needing some time until the board stays configured and not is resetting itself/reconfiguring. Maybe there is still some issue there regarding timing? Or I need to increase/add additional delay between accessing driver 0 and 1 (basically it got its own PCB and both drivers are just some centimeters away with the UART connection running over the PCB by a very short distance).

Included is the source code I tested and the serial output, thanks a lot already :)

````
#include <TMC2209.h>

//SerialPIO(txpin, rxpin, fifosize (32 bytes default)) 
SerialPIO Serial3(2,3);

//HardwareSerial & serial_stream = Serial3;
//const int DELAY = 3000;
const int DELAY = 500;

// Instantiate the two TMC2209
TMC2209 stepper_driver_0;
const TMC2209::SerialAddress SERIAL_ADDRESS_0 = TMC2209::SERIAL_ADDRESS_0;
TMC2209 stepper_driver_1;
const TMC2209::SerialAddress SERIAL_ADDRESS_1 = TMC2209::SERIAL_ADDRESS_1;
const uint8_t REPLY_DELAY = 4;
const long SERIAL_BAUD_RATE = 115200;

void setup()
{
  Serial.begin(SERIAL_BAUD_RATE);
  Serial.ignoreFlowControl(true);
  // TMC2209::SERIAL_ADDRESS_0 is used by default if not specified
  stepper_driver_0.setup(Serial3,SERIAL_BAUD_RATE,SERIAL_ADDRESS_0);
  stepper_driver_0.setReplyDelay(REPLY_DELAY);
  stepper_driver_1.setup(Serial3,SERIAL_BAUD_RATE,SERIAL_ADDRESS_1);
  stepper_driver_1.setReplyDelay(REPLY_DELAY);  
}

void loop()
{

  if (stepper_driver_0.isSetupAndCommunicating())
  {
    Serial.println("0 Stepper driver is setup and communicating!");
    Serial.println("0 Try turning driver power off to see what happens.");
  }
  else if (stepper_driver_0.isCommunicatingButNotSetup())
  {
    Serial.println("0 Stepper driver is communicating but not setup!");
    Serial.println("0 Running setup again...");
    //stepper_driver.setup(serial_stream);
    stepper_driver_0.setup(Serial3,SERIAL_BAUD_RATE,SERIAL_ADDRESS_0);
    //stepper_driver_0.setReplyDelay(REPLY_DELAY);  
  }
  else
  {
    Serial.println("0 Stepper driver is not communicating!");
    Serial.println("0 Try turning driver power on to see what happens.");
  }
  Serial.println();

  
  if (stepper_driver_1.isSetupAndCommunicating())
  {
    Serial.println("1 Stepper driver is setup and communicating!");
    Serial.println("1 Try turning driver power off to see what happens.");
  }
  else if (stepper_driver_1.isCommunicatingButNotSetup())
  {
    Serial.println("1 Stepper driver is communicating but not setup!");
    Serial.println("1 Running setup again...");
[... 516 lines cut ...]

0 Stepper driver is setup and communicating!
0 Try turning driver power off to see what happens.

1 Stepper driver is setup and communicating!
1 Try turning driver power off to see what happens.

0 Stepper driver is setup and communicating!
0 Try turning driver power off to see what happens.

1 Stepper driver is setup and communicating!
1 Try turning driver power off to see what happens.

0 Stepper driver is setup and communicating!
0 Try turning driver power off to see what happens.

1 Stepper driver is setup and communicating!
1 Try turning driver power off to see what happens.

0 Stepper driver is setup and communicating!
0 Try turning driver power off to see what happens.

1 Stepper driver is setup and communicating!
1 Try turning driver power off to see what happens.
````

nmaas87, 2023-10-27:
Thanks for the confirmation and updating the library.

Yes, I will keep an eye open and if I find the issue or get some more details/can pin it down i'll let you know :)

nmaas87, 2023-11-01:
So I did some additional testing. One problem I could quickly pin down and resolve by shortening the wire run for the UART, which helped a lot.

However, I still saw that issue that - at random - some drivers would just "toggle" all the time, e.g. being setup and re-setup all the time. I saw that this issue even got worse when I lowered the communication speed (even with increasing the delay). In the end I picked 250000 BAUD and this seemed to do the trick. Upon a toggle of the driver power, the drivers will be reset and re-setup. They might "toggle" some rounds, but after a short time they will settle and stay setup - I attach my testcode below.

I also found another issue that might be a problem with my understanding of the overall process/TMC2209 control: The TMC2209 is connected via DIR, STEP, ENABLE and UART lines to the RP2040.
Reason is that I want to use your library to just do two things:
a) Change the microstepping on one pair of drivers to "full step" (eg 2^0) to get maximium torque
b) Use the "parking" features, e.g. short-circuit the motorcoils to have the stepper stay in place more easily without the need to apply power.

For stepping the motor itself I am using https://github.com/laurb9/StepperDriver - which works very well using the BasicStepperDriver.

After I setup the TMC2209 before the BasicStepperDriver like this:
````
  stepper_driver_el.setup(SerialE,SERIAL_BAUD_RATE,SERIAL_ADDRESS_0);
  stepper_driver_el.setReplyDelay(REPLY_DELAY);
  stepper_driver_el.setMicrostepsPerStepPowerOfTwo(0);
  stepper_driver_er.setup(SerialE,SERIAL_BAUD_RATE,SERIAL_ADDRESS_1);
  stepper_driver_er.setReplyDelay(REPLY_DELAY);
  stepper_driver_er.setMicrostepsPerStepPowerOfTwo(0);
  stepper_driver_el.enable();
  stepper_driver_er.enable();
````
and especially software enabled the TMC2209 via UART, the BasicStepperDriver could just normally enable/disable the driver via the hardware interface (at least it looks like it) and everything works as before I included the TMC2209 for the UART interface.

However - the issue I see is that - if I connect the UART interface, it seems that the TMC2209 react vastly different to the steps then when the driver is running standalone/without UART. It looks like its having a completly different power profile, not enough torque and overall is louder and weaker. Is there any way to load the same "defaults" the TMC2209 has as if its not operating in UART mode?

Thanks so much! :)

Added the test script for the communication issues

````
#include <TMC2209.h>

// This example will not work on Arduino boards without HardwareSerial ports,
// such as the Uno, Nano, and Mini.
//
// See this reference for more details:
// https://www.arduino.cc/reference/en/language/functions/communication/serial/

//SerialPIO(txpin, rxpin, fifosize (32 bytes default)) 
SerialPIO SerialE(27,26); //EL,ER
SerialPIO SerialA1(2,3); //A1
SerialPIO SerialA2(6,7); //A2

//HardwareSerial & serial_stream = Serial3;
//const int DELAY = 3000;
const int DELAY = 500;

const TMC2209::SerialAddress SERIAL_ADDRESS_0 = TMC2209::SERIAL_ADDRESS_0;
const TMC2209::SerialAddress SERIAL_ADDRESS_1 = TMC2209::SERIAL_ADDRESS_1;

// Instantiate the two TMC2209
TMC2209 stepper_driver_a1_0;
TMC2209 stepper_driver_a1_1;
TMC2209 stepper_driver_a2_0;
TMC2209 stepper_driver_a2_1;
TMC2209 stepper_driver_el;
TMC2209 stepper_driver_er;

const uint8_t REPLY_DELAY = 4;
//const uint8_t REPLY_DELAY = 8;
//const long SERIAL_BAUD_RATE = 115200;
const long SERIAL_BAUD_RATE = 250000;

void setup()
{
  Serial.begin(SERIAL_BAUD_RATE);
  Serial.ignoreFlowControl(true);
  // TMC2209::SERIAL_ADDRESS_0 is used by default if not specified
  stepper_driver_a1_0.setup(SerialA1,SERIAL_BAUD_RATE,SERIAL_ADDRESS_0);
[... 128 lines cut ...]

  if (stepper_driver_er.isSetupAndCommunicating())
  {
    Serial.println("E0-1 OK!");
    //Serial.println("E1 Stepper driver is setup and communicating!");
    //Serial.println("E1 Try turning driver power off to see what happens.");
  }
  else if (stepper_driver_er.isCommunicatingButNotSetup())
  {
    Serial.println("E0-1 SETUP RERUN!");
    //Serial.println("E1 Stepper driver is communicating but not setup!");
    //Serial.println("E1 Running setup again...");
  stepper_driver_er.setup(SerialE,SERIAL_BAUD_RATE,SERIAL_ADDRESS_1);
  //stepper_driver_er.setReplyDelay(REPLY_DELAY);
  }
  else
  {
    Serial.println("E1 Stepper driver is not communicating!");
    Serial.println("E1 Try turning driver power on to see what happens.");
  }
  Serial.println();
  
  delay(DELAY);
}
````

peterpolidoro, 2023-11-01:
Yes it does not surprise me that strange things happen when the UART baud rate gets too high or the cables too long. The datasheet says that the maximum baud rate should be fclk/16 so with a 12MHz clock you should be able to use 750k baud, but cables and multiple chips probably lower that maximum rate significantly. Maybe 250k is the max with your particular setup.

Yes the default values used by the chip are very different than the values used in UART mode with this library. You can change some of the values used during standalone by setting the configuration pins on the chip. I mostly use the chip in UART mode, though, so I have full control over it. There is a mechanism for saving new values that will get loaded as default during standalone power up, but I have never used it. Look up `OTP configuration memory` in the TMC2209 datasheet.

nmaas87, 2023-11-01:
Thanks Peter,
yes too long cables and too high baud rates will be an issue, absolutly. The funny issue in this regard however was that using lower settings (e.g. 115200 BAUD, 57600 or even 19200) resulted in much more issues - even if I increased the delay. Choosing 250000 BAUD was the BAUD rate I choose that got the setup running smoothly, which is funny - as I would think lower datarates should be more reliable (at least thinking about the physics ;)). On the other hand, the RP2040 clocks at 133 MHz, so there is enough headroom available. However I would need to look into the details of the PIO interfaces - maybe they got some sort of sweetspot (in regards to where their base frequency arrives from :)) - which I maybe hit with the 250000 BAUD.

Yes, I read about the OTP already, its a bit scary to have to know exactly what you want - however, it should be possible to just test the driver settings beforehand until I am sure that this is all ok.
Is there any documentation available on what the "standalone mode" and "UART mode" settings/default differ from? Thanks :)