janelia-arduino/TMC2209 #40: Can't set my own RX TX pins on ESP32 module

peetman, 2023-10-26:
I'm trying to use my ESP32 module, but when I set my own RX TX pins, the Serial Monitor does not work. This is what I've done in the code. The rest is equal to the Communication example.

```

HardwareSerial & serial_stream = Serial1;

const long SERIAL_BAUD_RATE = 115200;
const int DELAY = 3000;

const int RX_PIN = 3;
const int TX_PIN = 1;

// Instantiate TMC2209
TMC2209 stepper_driver;

void setup()
{
  Serial.begin(SERIAL_BAUD_RATE);

  //stepper_driver.setup(serial_stream);
  stepper_driver.setup(serial_stream, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_0, RX_PIN, TX_PIN);
```

peterpolidoro, 2023-10-27:
So everything works if you do not use alternate pins, but as soon as you specify the alternate pins on Serial1, the serial monitor stops working?

I would think that Serial and Serial1 should be independent so changing the pins on one should not affect the other. I have not used an ESP32, but no one else has reported this problem. Is there anything special about pin 1 or 3 that might cause an issue? Can you try different pins to see if it still breaks? Maybe RX_PIN and TX_PIN are defined as macros somewhere. You could try renaming and see if that helps.

peetman, 2023-10-30:
Sorry I was using pins that are not UART capable, also pin 1 and 3 are directly connected to the usb, which means they can't be used for external UART communication. I've changed to Serial 2 and swapped to pins 16 and 17 and works great now. Thanks!
I have an AZ delivery esp32 nodemcu.

closed 2023-10-30 by peetman