janelia-arduino/TMC2209 #41: SW enable not working and connection issue

alxkos, 2023-11-06:
Hey, trying to setup tmc2209 with nano and SW serial. 
First: SW enable not working. Motor not locking and not moving.
Grounding enable pin gets its kinda moving. but not as expected.
With this another weird problem:
these is a signal on TX (using scope) with enable pin grounded and test sketch motor like moving, but getting 
'
Stepper driver is not communicating!
Try turning driver power on to see what happens.
'
Using one pin connection with 1k res TX-RX
External power source. 

`#include <Arduino.h>
#include <SoftwareSerial.h>
#include "TMC2209.h"

TMC2209 motor;
// HardwareSerial & serial_strea = Serial1;

const int RX_PIN = PIN3;
const int TX_PIN = PIN4;
const int EnPin = PIN2;
SoftwareSerial soft_serial(RX_PIN, TX_PIN);
const long SERIAL_BAUD_RATE = 9600;

const int32_t RUN_VELOCITY = 1000;
const int32_t STOP_VELOCITY = 0;
const int RUN_DURATION = 2000;
const int STOP_DURATION = 1000;

const uint8_t RUN_CURRENT_PERCENT = 60;

bool invert_direction = false;

void setup()
{
    Serial.begin(115200);
    Serial.println("conencted");

    motor.setup(soft_serial);
    motor.setRunCurrent(RUN_CURRENT_PERCENT);
    motor.enableCoolStep();
    motor.enable();
    Serial.println(motor.getSettings().is_setup);

}

//main loop reading serial
void loop()
{

    motor.moveAtVelocity(STOP_VELOCITY);
    delay(STOP_DURATION);
    if (invert_direction)
    {
        motor.enableInverseMotorDirection();
    }
    else
    {
        motor.disableInverseMotorDirection();
    }
    invert_direction = not invert_direction;

    motor.moveAtVelocity(RUN_VELOCITY);

    delay(RUN_DURATION);

    if (motor.isSetupAndCommunicating())
    {
        Serial.println("Stepper driver is setup and communicating!");
        Serial.println("Try turning driver power off to see what happens.");
    }
    else
    {
        Serial.println("Stepper driver is not communicating!");
        Serial.println("Try turning driver power on to see what happens.");
    }
    Serial.println();
}

`

Please advice. Cant work with the module. 
Thanks

alxkos, 2023-11-06:
Hey, after some reading and some fight i was able to run it in hardware serial mode, but not in software mode. 
not communicating ( 
What it can be?
Using nano with rx, tx pins 2,3

alxkos, 2023-11-06:
Can i get a response using software serial?

peterpolidoro, 2023-11-06:
The issue is with the software serial library. You should still be able to set all of the driver parameters, but you cannot get responses back when using software serial.

alxkos, 2023-11-06:
so i can use stallGuard? (using diag pin)