janelia-arduino/TMC2209 #42: AutoTuning

danielschlingmeier, 2023-11-22:
Hey!
I played around with your lib (thank a lot for the great job btw!) and nearly fried my stepper motor. When reading again you mentioned at several point to reduce the max current, when necessary, and better use voltage control etc.

I think I found out the reason for this "safety issue" with the current: One has to activcate the internal current sensing first and AFTERWARDS activate the tuning/current control, otherwise the TMC2209 auto tuning does, as I guess, not get a proper current signal for the tuning and does some nonsense leading to far to high actuation, which then fries the stepper.

By doing so:
	stepper_driver.useInternalSenseResistors();
	stepper_driver.enableAutomaticCurrentScaling();

It work beautifully, no overheating anymore. Maybe you want to try it out and consider this. I guess activating the internal sense before sensing the request for autotuning to the TMC2209 would fix this.

Best regards
Daniel

peterpolidoro, 2023-11-27:
Oh that is a good idea! I am wondering how I can enforce this behavior within the library to make it safer. What happens if you disable the driver before setting the sense resistors and then reenable the driver?

danielschlingmeier, 2023-11-27:
Well, I guess it would be sufficient to integrate *".useInternalSenseResistors();" internally in the ".enableAutomaticCurrentScaling();" method.
Not?

s-light, 2025-06-23:
does this make sense if the board (iam using) has external sens resistors?
(the classics 0.11R)