janelia-arduino/TMC2209 #43: Step motor speed

veyselolgun, 2023-12-08:
Hello,

I am using [Marlin Firmware](https://marlinfw.org/) to control Nema-17 stepper motor. Stepper driver is TMC2209

Step motor moving 1 micrometer with microstepping. When I commando step motor 1 micrometer movement, Marlin says that it complete this movement 0.1 seconds. Actually I am wondering that, is this 0.1 sec. normal ?

I think 1 micrometer movement can be completed much less than 0.1 seconds ? 

Can you please give me some idea what is wrong on my side ? Is it possible to complete 1 micrometer in 10ms with step motor ?

Thank you

peterpolidoro, 2023-12-08:
Hi,

I have never used Marlin firmware so I am not sure how that works. 
It is possible that 0.1 is the lowest value that it can show and 
it really happens much faster. I do not know if it is feeding step 
and direction signals in from a microcontroller or what.

Do you have an oscilloscope? You could look at the step signals 
and verify how many happen per second and figure out how fast it 
is really going. You need to know the conversion from microsteps 
to micrometers of travel on your setup.

veysel ***@***.***> writes:

> 1.  (*) text/plain          ( ) text/html           
>
> Hello,
>
> I am using [Marlin Firmware](https://marlinfw.org/) to control 
> Nema-17 stepper motor. Stepper driver is TMC2209
>
> Step motor moving 1 micrometer with microstepping. When I 
> commando step motor 1
> micrometer movement, Marlin says that it complete this movement 
> 0.1 seconds.
> Actually I am wondering that, is this 0.1 sec. normal ?
>
> I think 1 micrometer movement can be completed much less than 
> 0.1 seconds ? 
>
> Can you please give me some idea what is wrong on my side ? Is 
> it possible to complete 1 micrometer in 10ms with step motor ?
>
> Thank you

veyselolgun, 2023-12-08:
Hi @peterpolidoro ,

First of all, thanks for the fast reply

I think step motor fast but Marlin give me 0.1 second. I did some test with marlin and I see that motor rotate fast but marlin say time is 0.1 sec.  maybe acceleration and deceleration take much time i do not know

Sorry I have no oscilloscope, Actually If I disable motors then marlin give me same time. And also If I give 50 micrometer movement time is about 0.1 seconds.

I just want learn that is step motor has or has not this capabilities ? or there are some problem at software control ? If i find the exact problem I can do radical change for example i can change Marlin if it is problem

Also can you test this on your side if it is possible ?  

Thanks

peterpolidoro, 2023-12-09:
The stepper driver is probably plenty fast enough as long as you have the proper voltage, current, and gear ratios. Sounds like you may have an issue with marlin not reporting time smaller than 0.1s.

veyselolgun, 2023-12-12:
Thank you @peterpolidoro, I agree with you, I think Marlin not report less than 0.1 seconds. I opened an issue on [here](https://github.com/MarlinFirmware/Marlin/issues/26496)