janelia-arduino/TMC2209 #45: STM32G0B1RE MCU issue with multiple drivers using one UART

aaron-hentschel, 2023-12-20:
Hi all, I've been attempting to configure two 2209 stepper drivers sharing a single UART on an STM32G0B1RE based board (using the stm32duino framework) and have come across an issue initialising both drivers.  

The code below is what I am testing with.  I can successfully communicate to stepper_driver1 fine, but not stepper_driver2, however if I change the order of the driver setup (put stepper_driver2 before stepper_driver1) in the setup function then stepper_driver2 communicates fine, but stepper_driver1 does not.  

In addition, both drivers communicate fine if I keep the drivers powered up, then flash the controller with stepper_driver1 setup before stepper_driver2, then reflash the controller with stepper_driver2 setup before stepper_driver1 (keeping the drivers powered up the whole time).

Any help would be greatly appreciated!

`#include <TMC2209.h>

#define UART_TX2   PA2
#define UART_RX2   PA3
#define UART_TX4  PC10
#define UART_RX4  PC11

HardwareSerial SerialCon(UART_RX2, UART_TX2);

HardwareSerial SerialDrv(UART_RX4, UART_TX4);
HardwareSerial & serial_stream = SerialDrv;

const long SERIAL_BAUD_RATE = 115200;
const int DELAY = 3000;

TMC2209 stepper_driver1;
TMC2209 stepper_driver2;

void setup() {
  SerialCon.begin(SERIAL_BAUD_RATE);

  stepper_driver1.setup(serial_stream, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_1);
  stepper_driver1.setReplyDelay(4);

  stepper_driver2.setup(serial_stream, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_2);
  stepper_driver2.setReplyDelay(4);
}

void stepper1check() {
  if (stepper_driver1.isSetupAndCommunicating()) {
    SerialCon.println("stepper_driver1 is setup and communicating!");
  } else if (stepper_driver1.isCommunicatingButNotSetup()) {
    SerialCon.println("stepper_driver1 is communicating but not setup! Running setup again...");
    stepper_driver1.setup(serial_stream, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_1);
  } else  {
    SerialCon.println("stepper_driver1 is not communicating!");
  }
  SerialCon.println();
  delay(DELAY);
}

void stepper2check() {
  if (stepper_driver2.isSetupAndCommunicating()) {
    SerialCon.println("stepper_driver2 is setup and communicating!");
  } else if (stepper_driver2.isCommunicatingButNotSetup()) {
    SerialCon.println("stepper_driver2 is communicating but not setup! Running setup again...");
    stepper_driver2.setup(serial_stream, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_2);
  } else  {
    SerialCon.println("stepper_driver2 is not communicating!");
  }
  SerialCon.println();
  delay(DELAY);
}

void loop() {
  stepper1check();
  stepper2check();
}
`

peterpolidoro, 2023-12-20:
What error message does it give you, that the second driver is not communicating? Do you have an oscilloscope? It would be interesting to look at the UART line and see what is happening with the signals from both drivers.

Can you try increasing the reply delay to see if that helps? The maximum reply delay is 15.

Are you going to control these drivers with step and direction signals? It is possible that unidirectional communication is working properly but there is interference in the bidirectional communication. You may be able to set values in each driver and then drive them successfully with the step and direction signals even with the error about it not communicating properly.

aaron-hentschel, 2023-12-20:
Sorry Peter, I should have mentioned in my post that the second driver continually reports as "communicating but not setup".  I have attempted to use a larger reply delay of 10 and 15 however that didn't help.  Controlling the drivers using step and direction signals works fine.  To confirm I have the hardware wired up correctly, I used an alternate library (the TMCStepper library) which shows bidirectional UART is working fine to both drivers.

peterpolidoro, 2023-12-21:
Oh, that is very useful information, thanks. I may have figured out the problem. I made a change and updated the version to 9.0.7. Is it possible for you to try this version and tell me if the problem still exists?

aaron-hentschel, 2023-12-22:
Hi Peter, I tried version 9.0.7 however the issue still remains.

peterpolidoro, 2023-12-22:
Can you please try using the alternate pins when setting up each driver and see if that helps? 

Like this:
```
void setup() {
  SerialCon.begin(SERIAL_BAUD_RATE);

  stepper_driver1.setup(serial_stream, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_1, UART_RX4, UART_TX4);
  stepper_driver1.setReplyDelay(4);

  stepper_driver2.setup(serial_stream, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_2, UART_RX4, UART_TX4);
  stepper_driver2.setReplyDelay(4);
}
```

aaron-hentschel, 2023-12-23:
Hi Peter, I have tried adding the pins as you suggested and received the following error:

[image]

peterpolidoro, 2023-12-23:
I will have to dig into the stm32duino code to see if I can figure out how to fix it for you. I might not have a chance to get to it until after the holidays, but I will keep you posted. Thanks for the updates.

image.png (view on web)

peterpolidoro, 2024-01-02:
I am looking at the stm32duino code and I do not see anything that should cause it to fail. There may be a subtle bug in the way they begin or end their hardware serial port, but I am not sure right now what I can do with this library to work around that.

tfranssen, 2024-12-12:
I have the same issue. I can only communicatie with driver with address 0