janelia-arduino/TMC2209 #47: How to set current limit?

ArhumJain, 2023-12-25:
This has been mentioned in several other issue but as far as I can tell no solution has been found. Other stepper libraries allow passing in an absolute current limit in milliamps given an Rsense value. The board I'm using has a 110mOhm Rsense value but the README current setting functions all take percentages resulting in a scaled value that scales - what? How do I know what current the driver is limiting to? I don't want to accidentally fry my stepper motor.

peterpolidoro, 2024-01-02:
Yes this is a good point. I have been wanting to add this feature for a while. The Rsense value may be the only piece of information that the user needs to provide to the library to make the library work with a specific hardware configuration, but I want to make sure that is true for every driver setting before adding this feature. Open loop control, for example, just does PWM of the source voltage, so the current through the motor windings is a function of the source voltage value, the percent on time, the resistance of the motor windings, and the rotational speed of the motor. Closed loop control uses the Rsense resistors to set the output, but the current through the motor windings might only work properly if the calibration sequence has been performed properly. Just adding a simple function to convert percent to current based on the Rsense value may be misleading if it does not set the current properly, leading to burned out motors as well.

Defozo, 2024-03-10:
My stepper has very little torque on a 100% current setting. What should I do to increase that?

StefanL38, 2024-04-02:
Your documentation is pretty good. Lot's of details with schematics, lot's of example-codes.

So far your library provides only this percentage value.
Can you at least add information with example values for everything.
I mean real numbers for stepper-motor specs 
saying stepper-motor coil-resistance: 5 Ohms
max current: 0.8 A
continual voltage: 4V

TMC2209 is set to "what do I know mode" then 
using RUN_CURRENT_PERCENT = 100
means:  THIS number "**NNNN**" of mA
or
as you are talking about open-loop control: 
is one possible mode of the TMC2209 "open-loop" which means the TMC2209 does **NOT** use the R_sense-resistors?

Well if this is the case your library is almost useless. Almost any modern application uses higher voltages and current-limiting through R_sense resistors. This is 50% of the features these stepper-drivers offer compared to simple H-bridges

peterpolidoro, 2024-04-05:
The TMC2209 uses the sense resistors in some modes and ignores them in other modes. The run current setting meaning changes slightly depending on the mode. You are free to use whatever modes you want. Calculating the actual motor current depends on multiple settings and the user's exact setup. Read the datasheet.

astorun, 2024-05-22:
Any update on this implementation? I wonder if we can issue explicit current definition for motors by using sense resistor value.

peterpolidoro, 2024-05-23:
Yes I plan on adding this. It may take a few weeks, though. I am thinking of doing a major reorganization of all of the code in this library and add requested features such as these.

ngergihun, 2025-02-10:
Would be super nice to have some help on this topic. I think many people would appreciate it since not burning your motor is a petty important topic! :D

s-light, 2025-06-23:
@peterpolidoro  do you already done this?
i found this `setRMSCurrent(uint16_t mA,  float rSense,  float holdMultiplier)` funciton in the source..
https://github.com/janelia-arduino/TMC2209/blob/main/src/TMC2209/TMC2209.cpp#L239

and we (*only*) need some documentation for this?