janelia-arduino/TMC2209 #51: Arduino Due - examples don’t work

Sommjo11, 2024-01-08:
After loading TestCommunication on Arduino Due the script does not compile.

SoftwareSerial.h missing.

if I modify the .h file at line 13 to force SOFTWARE_SERIAL_IMPLEMENTED to false the software compiles but nothing is printed to screen. 

The processor hangs during setup.

Any tips on how to workaround would be much appreciated .

peterpolidoro, 2024-01-09:
Interesting. I will add Arduino Due to the list of boards that do not have Software Serial.

Have you tried any other examples? I am seeing some warnings about "constrain" when I use an older Arduino IDE, but I do not know if that could cause the processor to hang up. Which version of the IDE are you using?

peterpolidoro, 2024-01-09:
I created a new version, 9.0.9, which adds Arduino Due to the list of boards without Software Serial and I also created a function to replace constrain to fix compiler warnings. These changes will probably not fix your issue, but can you try the new version and see what happens with several examples? Thanks!

Sommjo11, 2024-01-09:
Thanks. Will do so soon.Joseph

Schweyher, 2024-07-15:
Hello Peter,

I am trying to set up TMC2209 on Arduino DUE.
Therefore I have chosen Bidirectional Communication examples and found that the Arduino DUE freezes after:
stepper_driver.setup(serial_stream); 
is put into the code - control LED is not blinking, Serial Monitor has no communication.
I have also tried the other setup examples, which do not work.

Is there any solution to run multiple TMC2209 on the Arduino DUE hardware? 

Kind Regards
Holger Schweyher

peterpolidoro, 2024-07-15:
I have never tried an Arduino DUE, but I do not know of any reason 
why it should not work. It has multiple UART outputs so you should 
be able to easily run multiple TMC2209 chips.

How do you have it wired? Which TMC2209 board are you using? Can 
you send an example of the code you are testing?

Schweyher ***@***.***> writes:

> Hello Peter,
>
> I am trying to set up TMC2209 on Arduino DUE.
> Therefore I have chosen Bidirectional Communication examples and
> found that the Arduino DUE freezes after: [...]

Schweyher, 2024-07-16:
Hello Peter,

I am using: 3D Printer Accessories TMC2209 V4.0 Quiet Stepping Motor Driver 5.5-28V, purchased on Ebay.
The Arduino Due was in my stock and could be of Chinese origin.
The wiring is okay, I think the problem is in the UART communication during setup.

I have put a LED blinking command and text into the code to check the functionality visually and in the Serial Monitor.
It works until the command: stepper_driver.setup(serial_stream); is put into the code, thereafter the LED stays on and no message on the Serial Monitor.

Can you recommend another TMC2209 breakout board, which usually works?

Kind Regards
Holger Schweyher

ThorsAvenger, 2024-08-03:
I have a Due as well, and have the same issues. I think the problem lies within the constructor for the Serial object. The controller freezes during setup if you try to pass the HardwareSerial &Serial object, but it will continue if you create the serial object first using Serial.begin(BAUD_RATE); and then pas the Serial object in the setup. (In my case I use Serial1) 

Then the driver will communicate, but won't be able to properly setup. getSettings() gives back 1 on communication but 0 on setup. Hope this sheds some light on the situation, I don't really know how to fix it from here...

dmoimas, 2026-01-29:
Hi everybody,

I'm using this library on Arduino Due, driving 14 TMC2209 on same serial port, bidirectional conneciton, with analog switches on the UART bus.

I've modified the following:
1. setup function
```
#if !defined(ARDUINO_ARCH_RENESAS)
void TMC2209::setup(HardwareSerial & serial,
  long serial_baud_rate,
  SerialAddress serial_address)
{
  hardware_serial_ptr_ = &serial;
#ifndef ARDUINO_ARCH_SAM
  hardware_serial_ptr_->end();
#endif
  hardware_serial_ptr_->begin(serial_baud_rate);
  initialize(serial_baud_rate, serial_address);
}
#endif
```
as Arduino Due get stuck if you run end() on a serial port without running a begin() first.

2. read function
```
    auto crc = calculateCrc(read_reply_datagram, WRITE_READ_REPLY_DATAGRAM_SIZE);
    delay(1);
    if (crc == read_reply_datagram.crc)
```
adding a delay before leaving read functions or retrying, as TMC2209 is holding UART bus causing Arduino unable to transmit other data without being corrupted. If data is corrupted, you must wait a bit longer for TMC2209 to reset and wait for another sync nibble. Checking datagram counter/IFCNT after each write (even unidir datagram send) would be great, but this implies bidir comms.