janelia-arduino/TMC2209 #53: Connect Rx and Tx separately

Okscientist, 2024-01-22:
Hi,

First of all, many thanks for your great library. I just started using your library. My set-up is an Arduino Mega with a RAMPS 1.4 board and 4 TMC2209 stepper drivers. I aim to read the Stallguard values via UART using the Tx Rx channels 1, 2 and 3 for three out of four stepper motors (0 is used for communication with the computer). 

Is it possible to connect Rx with Tx and Tx with Rx from my driver to the RAMPS pins. In your documentation I read that Rx and Tx wires are combined to form one wire, but my TMC2209 has both a Tx and Rx pin just like my Arduino Mega with RAMPS 1.4 board...

Hopefully you can help me out!

peterpolidoro, 2024-01-22:
Do you have a schematic for the board?

The UART settings will depend on how exactly the TX and RX lines are wired up on the board.

Okscientist, 2024-01-22:
Yes, please take a look below.

TX1 and RX1 pins in the Arduino Mega are pins D18 and D19, available on the RAMPS board at the top right section (section for End Stops).

[image]

peterpolidoro, 2024-01-22:
Is there a schematic for how the D18 and D19 pins connect to the TMC2209 chip on the board? There is only a single serial pin on the TMC2209. I need to see how that single pin gets connected to the external connector pins on the board to know what settings you need in the firmware.

Okscientist, 2024-01-22:
The TMC2209 I use has two serial pins. It is a TMC 2209 from Big Tree Tech (see the two PDN pins). They both are pointing upwards and downwards (downwards in the RAMPS board and upwards available for me to connect cables to)

[image]

Regarding the D18 and D19 pins, I would connect them myself manually from the RAMPS board top right corner to the PDN pins on the TMC2209 that are pointing upwards...

Okscientist, 2024-01-22:
To add: here is the schematic of the RAMPS board. It assumes a A4988 is plugged in, but a TMC2209 is plugged in. I could cut the PDN pins going into the RAMPS board off to make sure it doesn’t get other signals or such. 
[image]

peterpolidoro, 2024-01-22:
Oh I see, so you are using the Bigtreetech board. That board does not actually have separate TX and RX pins.

See the [schematic](https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/blob/master/Schematic/TMC2209-V1.2.pdf) 

You can solder the jumpers on that board so the single UART line is shared with two pins. In order to separate the TX and RX line, you will still need to add a resistor between the shared UART pins. Then you can use this library's standard settings.

Okscientist, 2024-01-25:
I actually have the BigTreeTech v1.3, but I can't find a schematic for that version.
There are no jumpers to solder. However, I will follow the steps as described in the ReadMe and will post the results!

Okscientist, 2024-01-26:
Unfortunately, it didn't work out just yet. After an extra inspection, I figured out I had to solder together these two pads (indicated with an arrow) to enable UART.
Any thoughts?
I will post the results as soon as I have new results!
[image]

Okscientist, 2024-01-26:
Did some testing. 

Very simple code to just read the Stallguard values;
`#include <TMC2209.h>

// Instantiate TMC2209
TMC2209 stepper_driver;
HardwareSerial & serial_stream = Serial1;
const long SERIAL_BAUD_RATE = 115200;

//Settings for stepper drivers
#define EN 38

//Direction pin
#define X_DIR A1

//Step pin
#define X_STP A0

//Speed & distance X (horizontal) Z (pump holder) and Y (pumping system)
int delayTimeX = 200;  //Delay between each pause (uS)

//Function Stepper Motor for X axis only
void step_XYZ(boolean dir, byte dirPin, byte stepperPin, int steps, int delayTime) {

  digitalWrite(dirPin, dir);

  for (int i = 0; i < steps; i++) {

    digitalWrite(stepperPin, HIGH);

    delayMicroseconds(delayTime);

    digitalWrite(stepperPin, LOW);

    delayMicroseconds(delayTime);
  }
}

void setup() {

  Serial.begin(SERIAL_BAUD_RATE);
  stepper_driver.setup(serial_stream);
  stepper_driver.enable();

  //Configuring correct pinmodes for stepper motors/drivers
  pinMode(X_DIR, OUTPUT);
  pinMode(X_STP, OUTPUT);

  // Enable motor
  pinMode(EN, OUTPUT);
  digitalWrite(EN, LOW);

}

void loop() {
  //Read Stallguard values
  uint16_t stall_guard_result = stepper_driver.getStallGuardResult();
  Serial.print("stall_guard_result = ");
  Serial.println(stall_guard_result);
}
`

This is what my Serial monitor looks like:
[image]

Anyone any advice??

Okscientist, 2024-01-29:
Hi!

I have got it working. Wired the TMC2209 to the MEGA with the RX of the TMC2209 to D14 (TX3) of the mega with a 1k Ohm resistor in between. The TX of the TMC2209 is directly connected to D15 (RX3) of the mega. 

I used the Stallguard example script, and it worked!!

Now I'm trying to fine-tune the script. 
The problem is that if I put the slightest amount of load on the motor shaft, the motor does not turn any longer. Stallguard values jump to high and low numbers all the time, even when the motor power is disconnected.

Any advice??

The script:
```
#include <TMC2209.h>

// This example will not work on Arduino boards without HardwareSerial ports,
// such as the Uno, Nano, and Mini.
//
// See this reference for more details:
// https://www.arduino.cc/reference/en/language/functions/communication/serial/

HardwareSerial & serial_stream = Serial3;

const long SERIAL_BAUD_RATE = 115200;
const int DELAY = 1000;
// current values may need to be reduced to prevent overheating depending on
// specific motor and power supply voltage
const uint8_t RUN_CURRENT_PERCENT = 100;
const int32_t VELOCITY = 2000;
const uint8_t STALL_GUARD_THRESHOLD = 255;

// Instantiate TMC2209
TMC2209 stepper_driver;

void setup()
{
  pinMode(4, OUTPUT);
  digitalWrite(4, LOW);

  Serial.begin(SERIAL_BAUD_RATE);

  stepper_driver.setup(serial_stream);
  stepper_driver.setRunCurrent(RUN_CURRENT_PERCENT);
  stepper_driver.setMicrostepsPerStep(8);
  stepper_driver.setStallGuardThreshold(STALL_GUARD_THRESHOLD);
  stepper_driver.enable();
  stepper_driver.moveAtVelocity(VELOCITY);

}

void loop()
{
  if (not stepper_driver.isSetupAndCommunicating())
  {
    Serial.println("Stepper driver not setup and communicating!");
    return;
  }

  Serial.print("run_current_percent = ");
  Serial.println(RUN_CURRENT_PERCENT);

  Serial.print("stall_guard_threshold = ");
  Serial.println(STALL_GUARD_THRESHOLD);

  uint16_t stall_guard_result = stepper_driver.getStallGuardResult();
  Serial.print("stall_guard_result = ");
  Serial.println(stall_guard_result);

  Serial.println();
  delay(200);

}
```

closed 2024-02-23 by peterpolidoro