janelia-arduino/TMC2209 #56: Mention some alternatives to the TMC429 in the readme

Luro02, 2024-02-22:
When I search for the chip, I can only find the datasheets and your library https://github.com/janelia-arduino/TMC429, seems like you can't buy them easily.

peterpolidoro, 2024-02-22:
Digi-Key seems to have plenty of TMC429 available:

https://www.digikey.com/en/products/detail/analog-devices-inc-maxim-integrated/TMC429-LI-T/4384204

Luro02, 2024-02-22:
That seems to be only the chip, right? I assume most people are like me using this library for an arduino project and I don't have the skills to do bga soldering or order a pcb to mount it (which will be quite expensive).

peterpolidoro, 2024-02-22:
There is a breakout board for the chip as well:

https://www.analog.com/en/resources/evaluation-hardware-and-software/evaluation-boards-kits/tmc429-bob.html#eb-overview

For sale here:

https://www.digikey.com/en/products/detail/analog-devices-inc-maxim-integrated/TMC429-BOB/8593663?s=N4IgTCBcDaICoFkDCAWMBOAtAIQPLYAIQBdAXyA

closed 2024-05-08 by Luro02