janelia-arduino/TMC2209 #57: assgin alternate rx and tx pins on waveshare RP2040 boards

paddle-salted-fish, 2024-03-04:
The boards in Waveshare RP2040 series have two serials and more than four pairs of UART pin. Each serial has two or three pairs of alternate rx/tx pins that need to be specified before use. Could you add "alternate rx/tx pin" support for them similar to ESP32?

peterpolidoro, 2024-03-04:
You can always set the alternate pins before setting up the TMC2209, but I went ahead and added support for passing those alternate pins into the setup method like on the ESP32. I have not tested it on hardware, but it compiles. Can you please test it and let me know if it works? Thanks!

paddle-salted-fish, 2024-03-05:
Thanks for your reply. I will test it on the waveshare RP2040-zero board soon.

paddle-salted-fish, 2024-03-06:
Thanks a lot for adding RP2040 support in the lastest codes v9.1.0. I have tested the code "TestRP2040" on the waveshare RP2040-zero board with TMC2209 drivers. It worked well when a driver was assgined address from 0-3 in the bidirectional communication model. When it came to multiple drivers connected to the same UART. The serial had the same ouput as mentioned in the [issue #38](https://github.com/janelia-arduino/TMC2209/issues/38). 
Here's my testing code. 
```
#include <TMC2209.h>

// set UART on Serial1 and TX/RX pin
SerialUART & serial_stream = Serial1;
const int RX_PIN = 1;
const int TX_PIN = 0;
// set serial response delay
const uint8_t REPLY_DELAY = 4;
// set Stepper EN pin
const uint8_t HARDWARE_ENABLE_PIN = 29;
// set the baudrate
const unsigned long SERIAL_BAUD_RATE = 115200;
const int DELAY = 3000;

TMC2209 steppers[3];
TMC2209::SerialAddress SerialAddresses[3] = {TMC2209::SERIAL_ADDRESS_0, TMC2209::SERIAL_ADDRESS_1, 
 									TMC2209::SERIAL_ADDRESS_2};

void setup() {
	// set UART pin and start serial
	for (uint8_t i = 0; i < 3; i++)
	{
		steppers[i].setup(serial_stream, SERIAL_BAUD_RATE, SerialAddresses[i], RX_PIN, TX_PIN);
		steppers[i].setReplyDelay(REPLY_DELAY);
		steppers[i].setHardwareEnablePin(HARDWARE_ENABLE_PIN);
		steppers[i].enable();
	}
}

void loop()
{
	for (uint8_t i = 0; i < 3; i++)
	{
		if (steppers[i].isSetupAndCommunicating())
		{
			Serial.print("Stepper driver ");
			Serial.print(i);
			Serial.println(" is setup and communicating!");
			Serial.println("Try turning driver power off to see what happens.");
		}
		else if (steppers[i].isCommunicatingButNotSetup())
		{
			Serial.print("Stepper driver ");
			Serial.print(i);
			Serial.println(" is communicating but not setup!");
			Serial.println("Running setup again...");
			steppers[i].setup(serial_stream);
		}
		else
		{
			Serial.print("Stepper driver ");
			Serial.print(i);
			Serial.println(" is not communicating!");
			Serial.println("Try turning driver power on to see what happens.");
		}
		Serial.println();
		delay(DELAY);
	}
}

```