janelia-arduino/TMC2209 #58: High level interface for enabling StallGuard DIAG pin to fire

cdaringe, 2024-03-10:
Thanks for this lib--it's awesome!

# Problem

Flashing the stallguard example works using UART registers, but the DIAG pin does not go hot, yielding perhaps an incomplete SG impl.

# Discussion

I'm after UART control, but I want very responsive SG detection. The DIAG pin has a few constraints on when it activates (https://www.analog.com/media/en/technical-documentation/data-sheets/TMC2209_datasheet_rev1.08.pdf). I'm still working through _why_ the pin doesn't rise even when this lib reports a SG state from SGRESULT.

peterpolidoro, 2024-03-11:
Interesting, I have not tested the DIAG pin output, but it would be great to figure out how to make this work.

The datasheet says:
DIAG is pulsed by StallGuard when SG_RESULT falls below SGTHRS. It is only enabled in StealthChop
mode, and when TCOOLTHRS ≥ TSTEP > TPWMTHRS

One or more of those conditions must not be true. I will have to experiment with this when I have some time. If you figure it out in the mean time, please let me know!

cdaringe, 2024-03-12:
I have a sneaking suspicion there may be something conflicting also with
the enable pin. On my board, the enable pin shares a trace with the
diagnostic pin, only because I have it slammed into a breadboard :-)

Sent from Gmail Mobile

cdaringe, 2024-03-18:
My `TSTEP` & `TCOOLTHRS` were within a percent of one another. I put a >=1% gap on them, and it triggers now. 😄 . Interesting that SW detection could trigger but not HW. Who knows, maybe weird analog digital conversion malarkey.

closed 2024-03-18 by cdaringe