janelia-arduino/TMC2209 #59: Bigtree and M4 Express; communication not working

Oegtsgeest, 2024-03-12:
Hi, 
My setup with an Adafruit M4-expressboard with a bigtree TMC2209 works when I use serial1.
However I need the serial1 fo a GPS shield, so I created Serial2 by using sercom code.
This created serialport on A3 (Rx) and A2 (Tx)  is tested seperately and works.

But I can not get it to work with the TMC2209. The biderectional testcommunication program fails.
Probably it has to do that I need an alternate Rx and Tx? But I do not understand then why the  HardwareSerial & serial_stream = Serial2 statement does not work.
Would it be enough if I change all the ESP32 words for  for __SAMD51__ in the TMC2209.h and TMC2209.cpp?
This before I start messing with the library. :-)
Please find below the code of the program.
Help is appreciated.
Joost
--------------
#include <Arduino.h>
#include <TMC2209.h>
#include "wiring_private.h"

// Serial2 on SERCOM4

#define PIN_SERIAL2_RX       A3  // RX
#define PIN_SERIAL2_TX       A2  // Tx
#define PAD_SERIAL2_RX       (SERCOM_RX_PAD_1)
#define PAD_SERIAL2_TX       (UART_TX_PAD_0)
const long SERIAL_BAUD_RATE = 115200;

Uart Serial2(&sercom4, PIN_SERIAL2_RX, PIN_SERIAL2_TX, PAD_SERIAL2_RX, PAD_SERIAL2_TX);

// Instantiate TMC2209
TMC2209 stepper_driver;
HardwareSerial & serial_stream = Serial2;    // this connects the Rx and Tx pins to the libary

void SERCOM4_0_Handler()   // required to get sercom4 working
{
  Serial2.IrqHandler();
}
void SERCOM4_1_Handler()
{
  Serial2.IrqHandler();
}
void SERCOM4_2_Handler()
{
  Serial2.IrqHandler();
}
void SERCOM4_3_Handler()
{
  Serial2.IrqHandler();
}

void setup() 
{
Serial.begin(115200);
while (!Serial) {} 
Serial.println("Begin set-up");
Serial2.begin(115200);
// Assign pins 12 &amp; 13 SERCOM functionality
pinPeripheral(PIN_SERIAL2_RX, PIO_SERCOM_ALT);
pinPeripheral(PIN_SERIAL2_TX, PIO_SERCOM_ALT);

stepper_driver.setup(serial_stream); 
Serial.println("Set-up completed"); 
}

void loop()
{
 
  Serial.println("start loop");
  
  if (stepper_driver.isSetupAndCommunicating())
  {
    Serial.println("Stepper driver is setup and communicating!");
    Serial.println("Try turning driver power off to see what happens.");
  }
  else if (stepper_driver.isCommunicatingButNotSetup())
  {
    Serial.println("Stepper driver is communicating but not setup!");
    Serial.println("Running setup again...");
    stepper_driver.setup(serial_stream);
  }
  else
  {
    Serial.println("Stepper driver is not communicating!");
    Serial.println("Try turning driver power on to see what happens.");
  }
  Serial.println();
  stepper_driver.getVersion();
  stepper_driver.getSettings();
  delay(2000);
}

peterpolidoro, 2024-03-14:
How does it fail exactly? Your code seems to compile. Do you mean that when you run it you are unable to communicate with the TMC2209? Do you have an oscilloscope? Can you look at the serial lines and see if it is sending the messages properly?

Oegtsgeest, 2024-03-14:
Thx for replying promptly.
I did change in th TMC2209.h and .cpp files as follows  #if defined ESP32 || defined __SAMD51__
And yes I have an oscilloscope but first i need to get the error solved.
But at compiling I get following error:

d:\......\Arduino\Sketches\libraries\TMC2209\src\TMC2209\TMC2209.cpp: In member function 'void TMC2209::setup(HardwareSerial&, long int, TMC2209::SerialAddress, int16_t, int16_t)':
d:\......\Arduino\Sketches\libraries\TMC2209\src\TMC2209\TMC2209.cpp:32:36: error: no matching function for call to 'HardwareSerial::end(bool)'
   32 |     hardware_serial_ptr_->end(false);
      |                                    ^
In file included from C:\Users\...\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.13\cores\arduino/Arduino.h:77,
                 from d:\.......\Arduino\Sketches\libraries\TMC2209\src/TMC2209.h:10,
                 from d:\.......\Arduino\Sketches\libraries\TMC2209\src\TMC2209\TMC2209.cpp:7:
C:\Users\.....\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.13\cores\arduino/HardwareSerial.h:72:18: note: candidate: 'virtual void HardwareSerial::end()'
   72 |     virtual void end() {}
      |                  ^~~
C:\Users\.....\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.13\cores\arduino/HardwareSerial.h:72:18: note:   candidate expects 0 arguments, 1 provided
d:\.........\Arduino\Sketches\libraries\TMC2209\src\TMC2209\TMC2209.cpp:38:97: error: no matching function for call to 'HardwareSerial::begin(long int&, long unsigned int, int16_t&, int16_t&)'
   38 |     hardware_serial_ptr_->begin(serial_baud_rate, SERIAL_8N1, alternate_rx_pin, alternate_tx_pin);
      |                                                                                                 ^
In file included from C:\Users\.....AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.13\cores\arduino/Arduino.h:77,
                 from d:\........\Arduino\Sketches\libraries\TMC2209\src/TMC2209.h:10,
                 from d:\........\Arduino\Sketches\libraries\TMC2209\src\TMC2209\TMC2209.cpp:7:
C:\Users\........\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.13\cores\arduino/HardwareSerial.h:70:18: note: candidate: 'virtual void HardwareSerial::begin(long unsigned int)'
   70 |     virtual void begin(unsigned long) {}
      |                  ^~~~~
C:\Users\........\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.13\cores\arduino/HardwareSerial.h:70:18: note:   candidate expects 1 argument, 4 provided
C:\Users\........\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.13\cores\arduino/HardwareSerial.h:71:18: note: candidate: 'virtual void HardwareSerial::begin(long unsigned int, uint16_t)'
   71 |     virtual void begin(unsigned long, uint16_t) {}
      |                  ^~~~~
C:\Users\jcgel\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.7.13\cores\arduino/HardwareSerial.h:71:18: note:   candidate expects 2 arguments, 4 provided

exit status 1

Compilation error: exit status 1

peterpolidoro, 2024-03-14:
What happen when you run your code with the unmodified TMC2209 library files? You should be already setting the alternate pins in your code, you should not need to do that inside the TMC2209 library code as well.

Oegtsgeest, 2024-03-14:
With stepper_driver.setup(serial_stream);  unmodified library ;  serial2,
I dont see  signal on the scope (stays LOW).

peterpolidoro, 2024-03-14:
I will have to read about sercom to know more about how to fix it.