janelia-arduino/TMC2209 #61: Bidirectional communication issue with Leonardo and TMC2209 with separate TX/RX pins

surei01, 2024-03-15:
Hi,
I am trying to get the bidirectional communication to work with a TMC2209 (GEEETECH) that has distinct pins for Tx and Rx and no single UART pin. 
I am connecting it to a Leonardo board, that has a hardware serial port (Serial1) on pins 0 and 1. I can control the driver via the hardware serial, so unidirectional communication from Leonardo to the driver is working, but I cannot read anything back. The examples for bidirectional communication always give me "Stepper driver not setup and communication".
Any ideas on what could be the cause? Is there any special wiring needed for a 2-wire serial connection? (other than connecting Tx with Rx and vice vers)

Any help is appreciated!
Best regards
Reiner

peterpolidoro, 2024-03-15:
Do you have a link to the schematic? It depends on how the Tx and Rx pins are wired exactly.

surei01, 2024-03-15:
unfortunately not, can't find one on the internet. It is a Geeetech TMS2209 v0.1. So far I only have fotos, but no schematic. Still googling for it.

surei01, 2024-03-15:
I do measure 1k between the TX and RX pins on the driver, interesting

surei01, 2024-03-15:
wow, they labelled the pins wrong on the board! Figured, since they are connected by 1k, which is how to connect the single wire interface, I might just try connecting TX to TX and RX to RX, and voila! It works like a charme, up to full 500k baudrate.

So, we can close this issue, and thanks for your help and thanks for the library!

peterpolidoro, 2024-03-15:
That is great!

closed 2024-03-15 by peterpolidoro