janelia-arduino/TMC2209 #63: How to get index pin pulsing?

Oegtsgeest, 2024-03-28:
Hi,
My configuration is Bigtree TMC2209.
The datasheet talks about a possibility of pulsing the index pin for each step. But I believe it is set now differently, because I see no output on the scope.
I need use of the index pin to get posistion while using internal pulsgenerator.
How do I get the index pulsing in each step?
Your help is appreciated.

aagum-bae, 2024-05-15:
Hi, set the bit `index_step` to `1` in the `GCONF` register

Oegtsgeest, 2024-05-15:
Hi, Thank you. Figured out this in the meantime too. But the value should be 0. The library does not support it. Therefore added in library as below. It works by calling e.g in the setup: stepper_driver.useIndexPin();  !
*   Added public function call: useIndexPin() in TMC2209.h 
*   and added in TMC2209 ccp :
*     void TMC2209::useIndexPin()         // function added 
*     {
*        global_config_.index_otpw = 0;   // the original value = 1
*        global_config_.index_step  = 0;   // the original value = 1
*        writeStoredGlobalConfig();
*      }

aagum-bae, 2024-05-16:
Awesome, the datasheet is confusing when it comes to when the index pin pulses, if you have gotten the index pin to pulse, can you tell me if it pulses for each micro step, or each fullstep, or if it pulses for 4 microsteps or if it pulses for 4 full steps, thanks!

Oegtsgeest, 2024-05-16:
Well, I did get it to work with the additions to the library as shown above. It gives a pulse per 4 full steps. See page 66 of the datasheet. (rev 1.09)

aagum-bae, 2024-05-17:
Hi the datasheet say the following:

> The index output allows precise detection of the microstep position within one electrical wave, i.e.,
> within a range of four fullsteps. With this, homing accuracy and reproducibility can be enhanced to
> microstep accuracy, even when using an inexpensive home switch.

How is it possible to detect microstep position if it guves one pulse per 4 fullsteps?

CBarninTeck, 2024-05-17:
if you set index_step to 1, The index pin will output the internal pulse generator signal, When it says it toggles on each step it will pull the pin to 2.5V, Then on the next step pulse it will pull it to -2.5V, on a scope it will show a PWM signal with half the Frequency of the internal pulse generator at 50% duty cycle (as one full wave is 2 pulses).  it looks something like this
(Yellow is current, Green is voltage. )
[image]

If you do it slow enough you will see the pulse then the voltage will drop (or rise) slowly to 0V, fast enough and you get the PWM. 

if you want only positive pulse you'll need a Precision Full-Wave Rectifier and something to manipulate the PWM to something more pulse looking or it will just be 2.5V all the time as whatever is behind the index pin drains slowly. 

I have tried this on a breadboard and it gives me what is expected just waiting for the opamps to be delivered to I can try the rest of the circuit 
[image]

im using this in the .h file 

```c++
  void EnableIndexOutput();
  void disableIndexOutput();
``` 
and this in the .ccp file
```c++
void TMC2209::EnableIndexOutput()
{
  global_config_.index_step = 1;
  writeStoredGlobalConfig();
}

void TMC2209::disableIndexOutput()
{
  global_config_.index_step = 0;
  writeStoredGlobalConfig();
}
``` 

And calling it in my code l like so.

```c++
  Rightasention.setup(Driveruart,9600,TMC2209::SERIAL_ADDRESS_0,16,17); 
  Rightasention.EnableIndexOutput();
``` 
I'm using a ESP32 with platform io.

aagum-bae, 2024-05-24:
Hi, first of all thanks for such a detailed reply.

> When it says it toggles on each step it will pull the pin to 2.5V, Then on the next step pulse it will pull it to -2.5V,

Does this mean, it toggles per each fullstep, or does it toggle per each microstep

CBarninTeck, 2024-05-24:
That will be dependent on what you set the index_step bit and the index_otpw bit too, if you want it to output a pulse at the starting position of each full step you need both bits set to 0.

If you want a pulse per step the you need only need index_step set to 1(as this overrides index_otpw). 

I haven't looked at the output of the index pin when both bits are set to 0 yet, I'll test it and edit this with the results for you.

aagum-bae, 2024-05-24:
Thanks a lot, what does the datasheet mean by 

> index_otpw
0: INDEX shows the first microstep position of sequencer

Shows first microstep position of sequencer?

CBarninTeck, 2024-05-29:
No worries , This part of the data sheet has the answers you are looking for: 
[image]

The index pin will pulse once per full step when in micro set mode (or in full step). That's what I'm getting from the data sheet any way.