janelia-arduino/TMC2209 #64: Stepper driver is not communicating! Try turning driver power on to see what happens.

aagum-bae, 2024-05-09:
I am using the example code of test communication, I am working with ESP32 and the TMC2209 driver, 

- I have connected the TMC2209 board with the ESP32 according to the wiring diagram
-  TX of driver to RX of ESP32, RX of driver to TX of ESP32 with 1k ohms resistor 
- the EN pin is connected to 3V therefore motor is disabled 
- and I have not connected any voltage source to the VM pin
- I have edited the example code by setting RX and TX pins of ESP32 to pins 16 and 17 respectively, I have attached the code below
- I have connected the VIO pin of the driver to the 3V pin of the ESP32
- I am getting the message "Stepper driver is not communicating! Try turning driver power on to see what happens." in the serial monitor
- What could be the issue?

`#include <TMC2209.h>

// This example will not work on Arduino boards without HardwareSerial ports,
// such as the Uno, Nano, and Mini.
//
// See this reference for more details:
// https://www.arduino.cc/reference/en/language/functions/communication/serial/

HardwareSerial & serial_stream = Serial2;

const long SERIAL_BAUD_RATE = 9600;
const int RX_PIN = 16;
const int TX_PIN = 17;
const int DELAY = 3000;

// Instantiate TMC2209
TMC2209 stepper_driver;

void setup()
{
  Serial.begin(SERIAL_BAUD_RATE);

  stepper_driver.setup(serial_stream, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_1, RX_PIN, TX_PIN);
}

void loop()
{
  if (stepper_driver.isSetupAndCommunicating())
  {
    Serial.println("Stepper driver is setup and communicating!");
    Serial.println("Try turning driver power off to see what happens.");
  }
  else if (stepper_driver.isCommunicatingButNotSetup())
  {
    Serial.println("Stepper driver is communicating but not setup!");
    Serial.println("Running setup again...");
    stepper_driver.setup(serial_stream);
  }
  else
  {
    Serial.println("Stepper driver is not communicating!");
    Serial.println("Try turning driver power on to see what happens.");
  }
  Serial.println();
  delay(DELAY);
}`

peterpolidoro, 2024-05-09:
Power might need to be connected to VM in order for it to communicate properly.

aagum-bae, 2024-05-09:
> Power might need to be connected to VM in order for it to communicate properly.

Hi Peter thanks for replying, I tried by connecting 10V to the VM pin, I still get same message

aagum-bae, 2024-05-10:
Hi all, fixed the problem, the FYSETC website for the driver breakout board I was using had given wrong motor wiring diagram, moreover the board has internal 1k resistor b/w RX and TX therefore, the TX of TMC must be connected to the TX of ESP and RX of TMC to RX of ESP.

FYSETC wasted 2 and 1/2 days worth of my time,

FYSETC website mentions: 
[image]
 However in reality the connectins are:
 **A1 and A2 are motor coil 1 and B1 and B2 are motor coil 2**
 
 This motor coil wiring along with proper connection of RX and TX and supplying voltage to VM pin fixed the problem.

closed 2024-05-10 by aagum-bae

pvginkel, 2024-12-14:
I just wanted to leave this here for the next person. I have a version 4.0 of the TMC2209 board and it does **not** have a 1K resistor on the board. It does have a TX and RX pin, but they connect to the chip without any resistance. This cost me some frustration but to be honest, the diagrams do say this:

Version 3.1:

[image]

Version 4.0:

[image]

The way to fix this is to take either of the pins on the board (doesn't matter which one), connect it to RX on the the MCU and connect the RX with a 1K resistor to the TX.

At least I have UART working now. Let's see what this baby can do 😊.