janelia-arduino/TMC2209 #65: Error during compilation of exmaple code for tetsing communication

aagum-bae, 2024-05-11:
Error message displayed:

c:\Users\user\Documents\Arduino\libraries\TMC2209\src\TMC2209\TMC2209.cpp: In member function 'void TMC2209::setup(HardwareSerial&, long int, TMC2209::SerialAddress, int16_t, int16_t)':
c:\Users\user\Documents\Arduino\libraries\TMC2209\src\TMC2209\TMC2209.cpp:32:36: error: no matching function for call to 'HardwareSerial::end(bool)'
     hardware_serial_ptr_->end(false);
                                    ^
In file included from C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.15\cores\esp32/Arduino.h:196,
                 from c:\Users\user\Documents\Arduino\libraries\TMC2209\src/TMC2209.h:10,
                 from c:\Users\user\Documents\Arduino\libraries\TMC2209\src\TMC2209\TMC2209.cpp:7:
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.15\cores\esp32/HardwareSerial.h:232:10: note: candidate: 'void HardwareSerial::end()'
     void end(void);
          ^~~
C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.15\cores\esp32/HardwareSerial.h:232:10: note:   candidate expects 0 arguments, 1 provided

exit status 1

Compilation error: exit status 1

For the example code:

#include <TMC2209.h>

// This example will not work on Arduino boards without HardwareSerial ports,
// such as the Uno, Nano, and Mini.
//
// See this reference for more details:
// https://www.arduino.cc/reference/en/language/functions/communication/serial/

HardwareSerial & serial_stream = Serial2;

const long SERIAL_BAUD_RATE = 115200;
const int RX_PIN = 16;
const int TX_PIN = 17;
const int DELAY = 3000;

// Instantiate TMC2209
TMC2209 stepper_driver;

void setup()
{
  Serial.begin(SERIAL_BAUD_RATE);
  stepper_driver.setup(serial_stream, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_0, RX_PIN, TX_PIN);
  
}

void loop()
{
  //stepper_driver.setup(serial_stream, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_0, RX_PIN, TX_PIN);
  if (stepper_driver.isSetupAndCommunicating())
  {
    Serial.println("Stepper driver is setup and communicating!");
    Serial.println("Try turning driver power off to see what happens.");
  }
  else if (stepper_driver.isCommunicatingButNotSetup())
  {
    Serial.println("Stepper driver is communicating but not setup!");
    Serial.println("Running setup again...");
    stepper_driver.setup(serial_stream, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_0, RX_PIN, TX_PIN);
  }
  else
  {
    Serial.println("Stepper driver is not communicating!");
    Serial.println("Try turning driver power on to see what happens.");
  }
  Serial.println();
  delay(DELAY);
}

Luro02, 2024-05-12:
That happens because you updated to the `3.0.0-rc1` version:
[image]

Can reproduce it by installing that version
```
d:\Users\Lucas\Documents\Arduino\libraries\TMC2209\src\TMC2209\TMC2209.cpp: In member function 'void TMC2209::setup(HardwareSerial&, long int, SerialAddress, int16_t, int16_t)':
d:\Users\Lucas\Documents\Arduino\libraries\TMC2209\src\TMC2209\TMC2209.cpp:32:30: error: no matching function for call to 'HardwareSerial::end(bool)'
   32 |     hardware_serial_ptr_->end(false);
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
In file included from C:\Users\Lucas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-rc1\cores\esp32/Arduino.h:197,
                 from d:\Users\Lucas\Documents\Arduino\libraries\TMC2209\src/TMC2209.h:10,
                 from d:\Users\Lucas\Documents\Arduino\libraries\TMC2209\src\TMC2209\TMC2209.cpp:7:
C:\Users\Lucas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-rc1\cores\esp32/HardwareSerial.h:255:10: note: candidate: 'void HardwareSerial::end()'
  255 |     void end(void);
      |          ^~~
C:\Users\Lucas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-rc1\cores\esp32/HardwareSerial.h:255:10: note:   candidate expects 0 arguments, 1 provided

exit status 1

Compilation error: exit status 1
```

Just downgrade to `3.0.0-alpha3` and it should work again.

Not sure if this is a bug in the esp32-arduino library or this project.

aagum-bae, 2024-05-12:
Thanks for the reply, however I am using 2.0.16, will try updating to 3.0.0 alpha 3

aagum-bae, 2024-05-14:
Hi all, problem solved, thanks @Luro02

peterpolidoro, 2024-05-14:
Do you have any recommendations on how I can change the library to best work with the various versions of ESP32? I will take a look at this when I have a chance.

Luro02, 2024-05-15:
One could adjust the code only for the esp32 3.x version;
```C
#if ESP_ARDUINO_VERSION_MAJOR >= 3
hardware_serial_ptr_->end();
#else
hardware_serial_ptr_->end(false);
#endif
```

Where is the `false` argument from and what does it do? Neither the ESP32 nor the Arduino Core AVR implementation has a `HardwareSerial::end(bool)` defined?

aagum-bae, 2024-05-15:
That works, but need to hit the reset button on the ESP32, figured it out after just messing around.

peterpolidoro, 2024-05-15:
I remember I needed to add end(false) to fix some bug, but I should have made a comment in the code about it because now I forget why I thought that was necessary. I think I will just replace that line with end() and people can upgrade to the newest esp32 3.x version if they run into any problems. Thanks!