janelia-arduino/TMC2209 #67: Stepper motor not wokring using STEP/DIR pins

aagum-bae, 2024-05-15:
Hi, this is my code, it worked fine for `stepper_driver.setMicrostepsPerStepPowerOfTwo(1);`  and `stepper_driver.setMicrostepsPerStepPowerOfTwo(0);`

But it is not working for `stepper_driver.setMicrostepsPerStepPowerOfTwo(2);`

Here is the code.

```
#include <TMC2209.h>

#define STEP_PIN            26

#define RX_PIN              16
#define TX_PIN              17
#define DIR_PIN             25
#define EN_PIN              13

HardwareSerial & serial_stream = Serial2;

const long SERIAL_BAUD_RATE = 115200;
const int DELAY = 2000;

bool flag = true;

const uint8_t RUN_CURRENT_PERCENT = 100;

TMC2209 stepper_driver;

void setup()
{
  pinMode(STEP_PIN, OUTPUT);
  pinMode(EN_PIN, OUTPUT);
  pinMode(DIR_PIN, OUTPUT);
  digitalWrite(EN_PIN, LOW);
  digitalWrite(DIR_PIN, HIGH);
  Serial.begin(SERIAL_BAUD_RATE);

  stepper_driver.setup(serial_stream, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_0, RX_PIN, TX_PIN);
  stepper_driver.moveUsingStepDirInterface();
  stepper_driver.setRunCurrent(RUN_CURRENT_PERCENT);
  stepper_driver.enable();
}

void loop()
{
    while(flag){

  
  stepper_driver.setMicrostepsPerStepPowerOfTwo(2); //takes value 0 to 8, it should map to 0-fullstep, 1-2uStep , 2-4ustep, 
  //                                                  3-8ustep, 4-16ustep, 5-32ustep, 6-64ustep, 7-128ustep, 8-256ustep
  
  // set i = 200 steps if microstep function has 0 as arguement for 360 degrees rotation
  // set 200*2 if 1 arguement
  // set 200*4 if 2 is arguement and so on
  
  for (uint16_t i = 800; i>0; i--) 
  {
    digitalWrite(STEP_PIN, HIGH);
    delay(1000);
    digitalWrite(STEP_PIN, LOW);
    delay(1000);
  }
  flag = false;
    }
}
```

Please help!

peterpolidoro, 2024-05-15:
I cannot think of any reason why the 2 case would be any different than the 1 or 0 case. What do you mean it is not working, the motor is not spinning or it is spinning incorrectly? What about the other values, 3-8?

aagum-bae, 2024-05-15:
Hi, thank you for the quick reply, the motor does not rotate, it just says in one place even for the other values 3-8 for the above code I have attached.

In the example code to set the microsteps, the motor doesn't rotate and stays in same position for values 0 and 1, however it rotates for the other values.

peterpolidoro, 2024-05-15:
What happens when you try the example in the directory: examples/BidirectionalCommunication/MicrostepsPerStep?

aagum-bae, 2024-05-15:
Motor stays stationary for the values 0 and 1, but rotates for the values 2-8

aagum-bae, 2024-05-16:
It has something to do with the delat between step pin going high and low, small delay value does not work with smaller steps

closed 2024-05-16 by aagum-bae