janelia-arduino/TMC2209 #68: Feature Request: Function aliases that use TMC2209 datasheet parameter names

LenStruttmann, 2024-05-25:
Admittedly, I am somewhat of a newbie, but one of the most difficult problems that I have using your wonderful library is tying the parameter names with your extended function names.  For example, using the TCM2209 datasheet I see that I must set TCOOLTHRS. But, how to do that?  I need to search through TMC2209.cpp and finally find that the function is setCoolStepDurationThreshold().  For me, my eyes just glaze over when I see such a big glob of characters and I find it nearly impossible to remember that setCoolStepDurationThreshold equates to TCOOLTHRS.  It would help me (and I suppose many others) if there was a function 
void TCOOLTHRS(uint32_t duration_threshold) to set the value.

Similarly, for values that can be read, a simple function like
uint32_t TSTEP() is, to me, much easier to tie into the datasheet than getInterstepDuration().

Thanks for all that you do.

aagum-bae, 2024-05-27:
Hi strut, if and until this is implemented, you can rename the functions yourelf and use.

LenStruttmann, 2024-05-27:
That is my plan, thanks!

peterpolidoro, 2024-05-28:
I am thinking of doing a pretty major rewrite, so I will definitely keep this in mind, thanks!