janelia-arduino/TMC2209 #69: Super low torque.

jkozniewski, 2024-05-26:
Hello - I'm testing the TMC2209 BIGTREETECH v 1.3 drivers without UART config and no matter which microstepping option I set via MS1/MS2 pins or what potentiometer value I set (tested max clockwise / counterclockwise positions) the motor has almost no torque I can stop the shaft by just gently grabbing / pressing it and any higher speed just makes the motor to vibrate instead of turning because of so low power.

I have tested the same stepper with A4988 driver and even with max microstepping it has much greater torque.

Then I have used this library and providing settings via UART  and managed to get more torque setting PWM offset / gradient manually to max values: 
```
stepper_driver.setPwmOffset(255);
stepper_driver.setPwmGradient(255);
```
although that makes stepper hot and much louder than with automatic settings:
```
stepper_driver.useInternalSenseResistors();
stepper_driver.enableAutomaticCurrentScaling();
stepper_driver.enableAutomaticGradientAdaptation();
```
that on the other hand give almost no torque like in standalone mode....

So is this default almost useless torque of TMC2209 BIGTREETECH v 1.3 something to be expected from those drivers or am I doing something terribly wrong ?

For the reference I'm using 200 steps / rev, 0.5A current 12v stepper, powered by 12v power supply rated for 3.0A
and as for microcontrollers I have used Arduino UNO and Teensy 4.1 (and Pico PI in standalone only mode as there seem to be no library for it unfortunately).

Anyone experienced same situation, any hints of how ro optimally setup the drivers either via this lib or any hardware-related means ? I have also posted an issue at Bigtreetech github - https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/issues/19...

aagum-bae, 2024-05-27:
Check #66

aagum-bae, 2024-05-28:
Hi, I think I might know what the problem is, see datasheet pg no 38 and pg no 39 `https://www.analog.com/media/en/technical-documentation/data-sheets/TMC2209_datasheet_rev1.09.pdf`

Especially pg no 39
For the automatic tuning to work properly, a tuning procedure is required first, morever are you setting the IRUN and IHOLD values properly in your code? 

I think with your full code it will be easier to help you

jkozniewski, 2024-05-29:
Hi - thanks for your feedback. In the end I've ditched UART mode, as for some reasons it started to work pretty decently in standalone mode (I guess I might did some stupid mistake with not setting ENABLE pin in the right state, curiously the motor runs regardless of wether ENABLE is high or low, which should not happen, and also in standalone mode it seems I can't really achieve true freewheeling behaviour - when disabled the the doing power is just a bit less then when enabled....)

Another thing - I have switched to 24v since just now I learned that I should not be worried about motor voltage rating and 12v can, and even should be driven by higher voltage - like 24v which substantially helped with the torque / speed...

Nonetheless in terms of UART and this library - i have encountered this "tuning procedure" but found none info of how should such procedure be performed - any insights regarding this matter  @peterpolidoro ? As for `IRUN` / `IHOLD` values - didn't find any examples of how to set them via library is it related to this example - https://github.com/janelia-arduino/TMC2209/blob/main/examples/UnidirectionalCommunication/Standstill/Standstill.ino ?
Seems that in standalone mode when disabled via ENABLE pin it defaults to something other than `FREEWHEELING` - my guess would be `NORMAL` or `BRAKING` ?

jkozniewski, 2024-05-29:
Thanks for clarification - all makes sense. Though it would be great to have some code example of actually performing the automated tuning procedure (even if it's not performed by default by the library for good reason you have mentioned) - I've read the docs and it's seems to be indeed described in detail but I'm not sure I understand everything to be able to translate it into code directly and small details might probably derail whole procedure making it hard to test / debug.

Luro02, 2024-05-30:
> stepper_driver.useInternalSenseResistors()

Try removing this line from your setup. I had a similar issue when I used the internal sense resistor.

Luro02, 2024-05-31:
That explains why my 2A stepper only draws around 0.5A, sometimes even less (not moving in that case) and recently it started drawing a lot more power [image]
[image]

I observed some weird behaviors that when I have the ESP32 connected via USB to the PC and use an external power supply the stepper is a lot slower than when I power the esp32 through a 12V -> 5V converter. All GND are connected through a ground plane on the pcb. I guess that this is related to me not doing the calibration procedure, resulting in random current draw.

---

From the datasheet I found this graphic, is this the calibration procedure for the current scaling? (This is for stealth chop? but the mentioned flags seem to suggest that it calibrates the current adaption?)
[image]

So to calibrate the stepper, you do the following:
1. Call the `enableAutomaticCurrentScaling()` and `enableAutomaticGradientAdaptation()` (not sure if you have to enable stealth chop explicitly as well?)
2. Then enable the driver through `stepper_driver.enable()` (this should charge the coils of the stepper motor, so it shouldn't move freely anymore?)
3. Just in case, move the stepper a single step and then move it back to it's original position
4. Wait for >130ms (I guess a `delay(300)` is enough?)
5. Not sure about the `PWM_GRAD_AUTO`, I guess this is the `enableAutomaticGradientAdaptation()`, so one can skip the medium speed movement calibration?

Is this it or did I miss something?

aagum-bae, 2024-06-02:
I have not tried the tuning procedure because for my application it is not required, however I have a few comments
 
1. Not required to enable stealthchop for tuning procedure
2. Yes it won't move freely, however if IHOLD is set to 0
3. Step not required if standstill current redution is not enabled, stand still current reduction is enabled when the PDN_UART pin is grounded
5. Yes

Will try it out myself when I get the time, for now I am trying to get the sensorless homing working.

Luro02, 2024-06-05:
I spent the last few days tweaking my code and reading the datasheet. Some things are still broken (the stepper is quite loud, even though I enabled StealthChop), but it is in a useable state.

Be careful about the current, I tried the code with a stepper motor that is rated for up to 2A (I am not even able to reach that current, likely because of the sense resistor on the pcb only working up to 1.77A). Therefore, it uses the entire range for the current from 0 to 100 (only the check_stall function would have to be adjusted and the constant for the run current).

Because you mentioned that you have low torque, I added my stall guard code that automatically increases/decreases the current when it does not have enough torque.

Not sure what your setup is, but I am pretty sure that you do not want to use the internal sense resistor. The pcbs with the chip on it, like this one
[image]
already have an external sense resistor on the pcb.
[image]
[image]

I get very low torque as well when I select the internal sense resistor. I haven't figured out how to configure it correctly, but the external sense resistor (selected by default) works fine.

<details>
  <summary>sketch.ino</summary>

```c
#include <AccelStepper.h>
#include <TMC2209.h>
#include <JC_Button.h>
#include <BlockNot.h>

const uint8_t DIR_PIN = 6;
const uint8_t STEP_PIN = 7;
const uint8_t EN_PIN = 0;

// ESP32-C6
#define RX_PIN 4  // = RX1
#define TX_PIN 5  // = TX1

// how many steps are required for a rotation (1.8° stepper requires 200)
const uint16_t FULL_STEPS_FOR_ROTATION = 200;
// how many microsteps should be between one step
const uint16_t MICROSTEPS_PER_STEP = 256;
// the number of full steps it should do per second (independent of the microsteps)
const uint16_t MAX_STEPS_PER_SECOND = FULL_STEPS_FOR_ROTATION * 20;
const uint16_t HALF_STEP_DURATION_MICROSECONDS = 10;

// For tuning the stepper has to move fast enough to generate enough back EMF
// and use the specified run current:
const uint16_t FAST_STEPS_PER_SECOND = FULL_STEPS_FOR_ROTATION * 4;

// How fast the stepper should accelerate to reach the specified speed.
//
// For example to reach a speed of 200 steps per second with an acceleration of 20,
// requires 10 seconds.
const float ACCELERATION_PER_STEP = 20.0;

const uint8_t RUN_CURRENT_PERCENT = 40;
const uint8_t HOLD_CURRENT_PERCENT = 20;

// This is the default value used by the TMC2209 library. After running the calibration while connected to a PC, update this with the value PWM_OFFSET_AUTO.
//
// User defined PWM amplitude offset (0-255) related to full
// motor current (CS_ACTUAL=31) in stand still. (Reset default=36)
//
// When using automatic scaling (pwm_autoscale=1) the
// value is used for initialization, only. The autoscale
// function starts with PWM_SCALE_AUTO=PWM_OFS and
// finds the required offset to yield the target current
// automatically.
//
// PWM_OFS = 0 will disable scaling down motor current
// below a motor specific lower measurement threshold.
// This setting should only be used under certain conditions,
// i.e. when the power supply voltage can vary up and down
// by a factor of two or more. It prevents the motor going
// out of regulation, but it also prevents power down below
// the regulation limit.
[... 239 lines cut ...]
    run_current = min(100, run_current + 2);
    stepper_driver.setRunCurrent(run_current);
    Serial.println("stepper has stalled");
  } else if (run_current > (uint8_t) (RUN_CURRENT_PERCENT / 2)) {
    run_current = max((uint8_t) (RUN_CURRENT_PERCENT / 2), (uint8_t) (run_current - 1));
    stepper_driver.setRunCurrent(run_current);
  }
}

void loop() {
  stepper.runSpeed();
  if (stall_timer.TRIGGERED) {
    check_stall();
  }
  if (stepper_communication_timer.TRIGGERED) {
    //debug_info();
    Serial.print("SG_RESULT = ");
    Serial.println(stepper_driver.getStallGuardResult());
  }
  if (led_blink_timer.TRIGGERED) {
    digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
  }
}
```
</details>