janelia-arduino/TMC2209 #70: Hardware UART no longer working on ESP32 boards due to Espressif updating from 2.X to 3.0

benpeart, 2024-06-06:
I recently started investigating using your library for a ESP32 based project I'm starting but could never get any of the UART based examples to work. After spending way more hours than I'd like to admit, I finally figured out why.

In November of 2023, the Arduino Espressif ESP32 boards package updated from 2.X to 3.0, and apparently the hardware UART code has changed as none of the examples run properly anymore. When I revert to version 2.0.17 of the boards library, they all work correctly.

All 3.0.0 version migration related issues are discussed [here](https://github.com/espressif/arduino-esp32/issues/8796) if you would like to look at them.

The UART specific migration issues are documented [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#uart-hardwareserial)

Thank you for all the work you have put into this library! Let me know if/how I can help.

peterpolidoro, 2024-06-06:
Hi! Which version of this library are you using? The last version update 9.1.1 should hopefully have fixed this issue. Were you using the latest version?

benpeart, 2024-06-06:
I'm using 9.1.1. Everything complies and 'runs' - it just never successfully communicates via UART. Using the Step/Dir/En pins works fine, just all serial communication/interfaces fail as if they aren't connected. I assumed forever that I just had the TX2/RX2 pins wired up wrong but after switching to the older board library they all work great.

[image]

```
C:\src\TMC2209>git log
commit 60e32c02b829d14033bfc18786ff6206d2dd4954 (HEAD -> main, tag: 9.1.1, origin/main, origin/HEAD)
Author: Peter Polidoro <peter@polidoro.io>
Date:   Wed May 15 11:04:33 2024 -0400

    end(false) -> end() to work with esp32 3.x versions.
```
Here is how I have things wired up but given it is working with the 2.x version of the Espressif board package I'm pretty sure things are correct:
[image]

LenStruttmann, 2024-06-06:
As an aside, I am working on a nearly identical set up and I am curious why you use two 22k resistors on the UART line rather than a single 1K resistor as shown on page 20 of the TMC2209 datasheet.  Obviously, it works for you, but I am curious as to why this was chosen.  Thanks

benpeart, 2024-06-06:
Sorry about the confusion there. I'm actually using a 1K resister on the TX line as specified on the datasheet. I was just being lazy and had never figured out how to choose the correct resister resistance in the Fritzing application (at some point I must have closed the Inspector window). I just googled it and fixed my laziness. :)

I had added a 10K pull down resister when trying to debug why UART communication wasn't working and observing via an oscilloscope that the line appears to be held high and I wasn't getting a big drop in voltage as the data was getting sent. I just removed the 10K pull down resister (now that I figured out that the issue was actually the 3.0 ESP32 board library) and it's still working correctly.

Here is an updated breadboard diagram with both issues fixed:

[image]

peterpolidoro, 2024-06-12:
So do you have any idea of what I need to change in this library in order to get it to work with 3.0?

benpeart, 2024-06-12:
I'm sorry but I really don't. I'd start by looking at the [UART specific migration issues](https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#uart-hardwareserial) to see if you are doing any of the things they have listed in the "functional changes" as those are more insidious than API changes which typically generate compiler errors.

benpeart, 2024-09-27:
I now understand the cause of this issue and have a simple workaround. I thought I'd add it here in case someone else runs into the same problem. 

The underlying change that caused the issue was listed in the 3.0.0 version migration related issues [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#id14).

>Default pins for some SoCs have been changed to avoid conflicts with other peripherals: * ESP32’s UART1 RX and TX pins are now GPIO26 and GPIO27, respectively; * ESP32’s UART2 RX and TX pins are now GPIO4 and GPIO25, respectively; * ESP32-S2’s UART1 RX and TX pins are now GPIO4 and GPIO5, respectively.

This means that GPIO pins for Serial and Serial2 have changed and as a result, they no longer match all the existing HW designs and all the example code, pinout diagrams and documentation out there on the internet are now wrong. It is an interesting choice that Espressif made but at least they made it fairly easy to work around.

The simplest solution is to set the pins used for Serial and Serial2 to be the pins that were used before this breaking change. This can done by explicitly setting the pins used to their old/default values when you initialize the serial port:

```
    Serial2.begin(115200, SERIAL_8N1, 16, 17); // Initialize HW UART drivers with the legacy RX/TX pins of 16/17
```

If you are using a library that calls begin on your behalf you can also call setPins() to change back to the old default pins and the pin assignments will be preserved when the library calls begin.

```
    Serial2.setPins(16, 17); // Initialize HW UART drivers with the legacy RX/TX pins of 16/17
```

I hope this can save someone else some frustration when they attempt to update their design to use the new Espressif libraries.

peterpolidoro, 2024-09-27:
Thank you! I will add this information to the documentation when I get a chance.

Can you confirm if it works when you call:

`TMC2209 stepper_driver;`
`stepper_driver.setup(Serial2, 115200, TMC2209::SERIAL_ADDRESS_0, 16, 17);`

girrrrrrr2, 2024-10-10:
it does not work

` *  Executing task in folder 2209: platformio run --target upload 

Processing upesy_wroom (platform: espressif32; board: upesy_wroom; framework: arduino)
------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/upesy_wroom.html
PLATFORM: Espressif 32 (6.8.1) > uPesy ESP32 Wroom DevKit
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TMC2209 @ 9.4.2
Building in release mode
Compiling .pio/build/upesy_wroom/src/main.cpp.o
Compiling .pio/build/upesy_wroom/FrameworkArduino/FirmwareMSC.cpp.o
Compiling .pio/build/upesy_wroom/FrameworkArduino/FunctionalInterrupt.cpp.o
src/main.cpp:21:1: error: 'stepper_driver' does not name a type
 stepper_driver.setup(Serial2, 115200, TMC2209::SERIAL_ADDRESS_0, 16, 17);
 ^~~~~~~~~~~~~~
*** [.pio/build/upesy_wroom/src/main.cpp.o] Error 1
================================== [FAILED] Took 4.32 seconds ==================================

 *  The terminal process "platformio 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. `
 
 and I am using the code for MoveAtVelocity.ino with the only changes being that added
 `stepper_driver.setup(Serial2, 115200, TMC2209::SERIAL_ADDRESS_0, 16, 17); `
 below the existing TMC2209 stepper_driver;

mfoste, 2024-12-04:
@peterpolidoro - I was running into the same issue as @benpeart. I have confirmed that calling setup() with the pin definition does allow the TMC2209 driver to communicate as expected using the Serial2 HW UART of my WROOM ESP32 board.

My exact setup:
`stepper_driver.setup(Serial2, 9600, TMC2209::SERIAL_ADDRESS_0, 16, 17);`