janelia-arduino/TMC2209 #72: How to set Microsteps with only having SoftwareSerial

Zchulski, 2024-06-15:
all the documentation and examples only point to being able to set the Microsteps if you have hardware serial.  Unfortunately my setup does not allow for using another but an arduino nano that doesn't have a hardware serial port.  How are we supposed to set the Microsteps if all the examples for setting it are only in the bi-directional communication section.  Can we still set the microsteps without having to call for a response from the TMC2209.

`#include <SoftwareSerial.h>
#include <TMCStepper.h>
#include <AccelStepper.h>

//Define Pins
//For LP:
#define UpPin A0
#define DownPin A1
//For DAC:
//#define UpPin A2
//#define DownPin A3
///===
#define EnInput 4
#define EnOutput 5
#define DirPin 2
#define StepPin 3

//Software Serial Setup Pin
#define SW_TX 7
#define SW_RX 8
#define DRIVER_ADDRESS 0b00
#define R_SENSE 0.11f

//TMCStepper Setup
SoftwareSerial UART(SW_RX, SW_TX);
TMC2209Stepper Driver(&UART, R_SENSE, DRIVER_ADDRESS);

//Accel Stepper Setup
AccelStepper Stepper(AccelStepper::DRIVER, StepPin, DirPin);

void setup() {
  pinMode(EnInput, INPUT);
  pinMode(EnOutput, OUTPUT);
  pinMode(DirPin, OUTPUT);
  pinMode(StepPin, OUTPUT);
  pinMode(SW_RX, INPUT);
  pinMode(SW_TX, OUTPUT);
  //Set EnPin Default High
  digitalWrite(EnOutput, HIGH);
  // Delay before Intialize
  delay(1000);

  // Intialize Driver if UART
  Driver.begin();
  Driver.toff(5);
  Driver.rms_current(900);

  // Set microsteps and print the value for debugging
  int desired_microsteps = 128; // Adjust this to your desired value
  Driver.microsteps(desired_microsteps);
  Serial.print("Microsteps set to: ");
  Serial.println(Driver.microsteps()); // This might not be available in all libraries, check documentation

  Driver.en_spreadCycle(false);
  Driver.pwm_autoscale(true);
  Stepper.setMaxSpeed(1000);
  Stepper.setAcceleration(200);
}

void loop() {
  digitalWrite(EnOutput, digitalRead(EnInput));

  int UpSpeed = analogRead(UpPin);
  int DownSpeed = analogRead(DownPin);

  if (UpSpeed > 40) {
    digitalWrite(DirPin, LOW);
    while (analogRead(UpPin) > 40) {
      UpSpeed = analogRead(UpPin);
      float speed = map(UpSpeed, 0, 1023, 0, 1000);
      Stepper.setSpeed(UpSpeed);
      Stepper.runSpeed();
    }
    Stepper.setSpeed(0);
  } else if (DownSpeed > 40) {
    digitalWrite(DirPin, HIGH);
    while (analogRead(DownPin) > 40) {
      DownSpeed = analogRead(DownPin);
      float speed = map(DownSpeed, 0, 1023, 0, 1000);
      Stepper.setSpeed(DownSpeed);
      Stepper.runSpeed();
    }
    Stepper.setSpeed(0);
  } else {
    Stepper.setSpeed(0);
    Stepper.runSpeed();
  }
}
`

I have tried using this and calling the driver to respond with the microsteps its using but it still always responds with 256.  If I set the current the driver correctly responds with the set current but it always responds with 256 when calling for what microstep setting its using.

peterpolidoro, 2024-06-21:
Try using:

Driver.setMicrostepsPerStep(desired_microsteps);

sepastian, 2024-10-01:
@Zchulski you are not using this library, you are using https://github.com/teemuatlut/TMCStepper.