janelia-arduino/TMC2209 #73: Help need: No move after UART setup

fraro, 2024-06-23:
I have an RP2040 connected to two 2209, with TX-only to both drivers (one tx of the RP20240 goes to both drivers).
The moveAtVelocity example works fine, so I guess the UART setup is correct.
And the motors are connected the right way.

But Step/Dir example does not work. After calling setup(), the hold current drops from 2x700mAh to 0.
If I don't set up the driver, moving the motors using the step/dir pins work fine.
I have the enable pin on the drivers pulled to GND, because I want to use sw control. I alway have the enable() as part of my code. 
I have connected  
- step/dir -> RP20240 GPIOs
- TX -> RP2020 serial
- EN -> GND
- A1/A2/B1/B2 -> motor
- Vm -> power supply
- Vio -> RP20240 board 3.3V

Any help appreciated!

peterpolidoro, 2024-06-24:
That is odd that the moveAtVelocity example works, but the StepAndDirection example does not. Which TMC2209 boards are you using? Are you setting the TX and RX pins before you call setup? Or are you passing the alternate RX and TX pins into the setup function?

LenStruttmann, 2024-06-24:
Do not forget that, even with the ENABLE pin connected to GND, you must still also enable the driver with a call to driver.enable()

fraro, 2024-06-24:
here's my setup:
```
  const uint8_t STEP_PIN = 14;
  const uint8_t DIR_PIN = 15;
  const long SERIAL_BAUD_RATE = 115200;
  const int RX_PIN = 9;
  const int TX_PIN = 8;
...
  stepper_driver.setup(Serial2, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_0, RX_PIN, TX_PIN);
  pinMode(STEP_PIN, OUTPUT);
  pinMode(DIR_PIN, OUTPUT);

  stepper_driver.enableCoolStep();
  stepper_driver.enable();
```
Hardware is a standard Pi Pico RP2020 board.
This is very odd, indeed. I guess there's just a stupid mistake, but...

fraro, 2024-06-24:
Driver board is the TMC2209 from here: https://de.aliexpress.com/item/1005005457405465.html 
I have tried to call the moveUsingStepDirInterface() before, but did not help.

I can try to solder the TX/RX variant to one TMC2209 instead, for reading the status from the driver. Does this make sense?

LenStruttmann, 2024-06-24:
For what it's worth, here are the other settings that I make for my current project.  Note, I am using BIGTREETECH boards.

stepper_driver->setRunCurrent(90); // RUN_CURRENT_PERCENT);
stepper_driver->enableAutomaticCurrentScaling(); // pwm_autoscale
stepper_driver->setCoolStepDurationThreshold(COOL_STEP_DURATION_THRESHOLD); // TCOOLTHRS = 2000
stepper_driver->setStealthChopDurationThreshold( STEALTH_CHOP_DURATION_THRESHOLD); // TPWMTHRS = 100
stepper_driver->setMicrostepsPerStepPowerOfTwo(8); 
stepper_driver->setHoldCurrent(0); // 0% 
stepper_driver->setStandstillMode(TMC2209::NORMAL);

peterpolidoro, 2024-06-25:
I do not see anything obviously wrong. Can you see that the step and direction pulses are working properly with an oscilloscope? Can you try connecting to only a single 2209? What voltage are you using for Vm? Can you show us your entire ino file?

fraro, 2024-06-25:
I think I found the reason (reading the 2209's status helped): The microsteps are 1/256 on init. From the example code step/dir 
  {
    digitalWrite(STEPB_PIN, !digitalRead(STEPB_PIN));
    delayMicroseconds(HALF_STEP_DURATION_MICROSECONDS);
  }

I expected the default to be 1/2. So my movements (number of steps) were just to small to notice.

(It's a camera pan/tilt head, where crashing into the end would stress the mechanics, so only small movements in the beginning)

Thank you both for the support and confirming the correct setup!

closed 2024-06-25 by fraro