janelia-arduino/TMC2209 #75: RP2040 PIO Support not available anymore since RP2040 update

nmaas87, 2024-06-30:
Dear all,
RP2040 posses PIO Blocks ( https://tutoduino.fr/en/pio-rp2040-en/ )which allow to generate e.g. 4 additional full-directional UARTs which are offloaded to the PIO blocks (see: https://arduino-pico.readthedocs.io/en/latest/piouart.html ).

In the past this was as easy in integration as creating an new PIO UART:
````
SerialPIO SerialE(27,26);
```` 
and then using it with TMC2209 like this:
````
stepper_driver_el.setup(SerialE,SERIAL_BAUD_RATE,SERIAL_ADDRESS_0);
````

however, this does not work anymore since the 410d73547a51521ff2d730928e1d7a6469f9c354 commit:

````
X:\sketchbook\libraries\TMC2209\src/TMC2209.h:43:27: note:   initializing argument 1 of 'void TMC2209::setup(SerialUART&, long int, SerialAddress, int16_t, int16_t)'
   43 |   void setup(SerialUART & serial,
      |              ~~~~~~~~~~~~~^~~~~~

cannot convert 'SerialPIO' to 'SerialUART&'
````

Any way to get this working again? Thank you! :)

peterpolidoro, 2024-07-01:
Good point! I just modified the code to hopefully add support for SerialPIO. Can you please try version 9.2.0 and see if it works for you? Thanks!

nmaas87, 2024-07-01:
Yes, thanks a lot, that worked! Can confirm works again! 👍🏻

closed 2024-07-01 by nmaas87