janelia-arduino/TMC2209 #76: Driver must be reset by removing power

SoupySoups, 2024-07-01:
This issue has been driving me crazy. I have a single driver board with an esp32 and seem to be having some weird behavior. Here is my schematic (sorry for the awful layout, I usually have time to do better)
[image]
There are a couple undesired behaviors that I have found:
1. Upon being stalled (on purpose by my hand) the motor will kinda just give up. There is lots of torque but it will die and not work again. Resetting does not help. The only thing that causes the motor to live again is to remove then apply motor power again.

Motor is being powered via a barrel jack, @24v with up to 8A at its disposal.

SoupySoups, 2024-07-01:
https://github.com/janelia-arduino/TMC2209/assets/101527636/30f8f5a3-decb-4f2e-a933-c618e4b8147f

Here is it in real life.

peterpolidoro, 2024-07-01:
I am not sure what is the problem exactly. Try putting a delay(4000); after stepper_driver.enable(); in the setup. Maybe the driver needs a little time to settle after stalling and then being commanded to stop before being commanded to move at a new velocity.

SoupySoups, 2024-07-01:
Same result. Is there any reason why it would give up? Not only is it stalling but the tmc stops outputting signals when it is stopped. I know this library comes with many of the safer parameters set, is there a chance that there is some setting set to protect the hardware?

peterpolidoro, 2024-07-01:
The driver might shut itself down if it is getting too hot. The datasheet also says: 

> A motor stall or a sudden change in the motor velocity may lead to the driver detecting a short circuit
>or to a state of automatic current regulation, from which it cannot recover. Clear the error flags and
>restart the motor from zero velocity to recover from this situation.

I will add a function to read the global status and another to clear the global status. Then on reset I will make it clear the global status by default. That might fix your problem, I will let you know when I update the version.

peterpolidoro, 2024-07-01:
Ok I have updated the library to version 9.4.0. This version automatically clears the drive error when it resets. You can also call clearDriveError() to clear it yourself to recover from a stall without resetting. Can you please try that version and see if it works for you?

SoupySoups, 2024-07-01:
Apologies for the delayed response. None of the temperature statuses have been triggered when the behavior occurs. Version 9.4.0 does not solve the problem. Calling any of the new functions within the loop does not change anything. Additionally, using the code
```cpp
  TMC2209::GlobalStatus gstat = stepper_driver.getGlobalStatus();

  Serial.println("GlobalStat:");
  Serial.print(" - reset: ");
  Serial.println(gstat.reset);
  Serial.print(" - drv_err: ");
  Serial.println(gstat.drv_err);
  Serial.print(" - reserved: ");
  Serial.println(gstat.reserved);
  Serial.print(" - uv_cp: ");
  Serial.println(gstat.uv_cp);
  ```
Reveals that no values change in the time that the motor is running vs the time in which it has stopped.

SoupySoups, 2024-07-01:
```
GlobalStat:
 - reset: 1
 - drv_err: 0
 - reserved: 69246976
 - uv_cp: 0
 ```

SoupySoups, 2024-07-01:
It seems to not be an issue when stealthchop is disabled but I cant get reliable operation without stealthchop.

peterpolidoro, 2024-07-02:
Only stealthchop, that is interesting. I am not sure what is happening, I will have to dig through the datasheet when I have more time. Is the speed you are jumping to very fast? The datasheet does say you may need to increment the speed in steps when using moveAtVelocity. I am not sure why it would work at first and then stop working after a stall though. Does the driver seemed disabled or is it still attempting to spin the motor? Other TMC drivers have an option to stop on stall detection, but I do not think the TMC2209 has this option.

SoupySoups, 2024-07-03:
no the speed is not fast at all, my accel function starts at 0. The driver is completely disabled, no attempt to spin.

SoupySoups, 2024-07-03:
At lower speeds it will even give up without any torque. It will move a little bit, a couple of steps, then give up.

dprotter, 2025-06-24:
Just throwing out that I'm having the same issue with an Arduino to TMC2209. If I cycle the arduino the driver does not work. if I THEN cycle the power to the driver, it works and I can alter microsteps etc.