janelia-arduino/TMC2209 #77: Potential Standstill Mode issues

nmaas87, 2024-07-05:
Dear all,
I tried the standstill mode example from the unidirectional examples.
Sadly, it seems like the standstill modes set have no effect.
Running the example switches through the modes while you cannot feel any change physically.
If I add the SettingsAndStatus example after each change, I can see that the standstill mode will always be at "1", which means freewheeling.
Is there some issue or am I doing something wrong?
Thanks a lot :)

peterpolidoro, 2024-07-15:
What is your setup? How do you have it wired? Do other examples work? Is your motor large or small?

nmaas87, 2025-01-02:
Hello Peter,
I got the issues solved.
The first problem was that I was using another library to control the stepping of the motor, meaning I had ENABLE, DIR and STEP connected to the driver. So the TMC2209.enable(); was not enough to turn the driver on, but I also needed to pull the ENBALE pin low.
The second issue is that I am using RP2040 with the PIO UART and the TMC2209 library was losing connection to the driver on lower connection speeds and reseting it all the time, interfering with the Standstill configuration. In the end I resolved this by setting the speed to at least 115200 BAUD and increased the default 32 bytes PIO UART FIFO to 64 bytes. Then it became stable, even if not running the PIO UART with speeds of 250000 or 500000 BAUD which were previously the only way I could get the library to work at all.
Thanks a lot

closed 2025-01-02 by nmaas87