janelia-arduino/TMC2209 #78: TMC2209 + RP2040 chip - no way to get it working

fryc88, 2024-07-14:
At the start i must say i'm total new with arduino ide, so i might be doing something wrong, however i'm trying to use 3d printer board for other purpose, but by all means i can't get it working.
There's RP2040 chip built in, 2x TMC2209, so i would assume tmc2209 lib for arduino ide. I'm not bothered about two, i'm not even bothered about using them in full uart mode, just normal step/dir would be fine too, but that way i couldn't get it working either
[https://github.com/FYSETC/FYSETC-ERB/blob/main/V1.0/hardware/Schematic.pdf](board schematics)

So from examples i've tried software serial, that wont even compile because of lack of soft serial library (?).
[image]
Any other options will give me similar results while trying to compile, all errors are related to lack of serial.
 
[image]

Could you tell me if that's related to RP2040 & you addon missing something, or am i missing something here?

peterpolidoro, 2024-07-14:
I have never used one of these boards, but I am curious to see if we can get it working.

I added an example examples/UnidirectionalCommunication/TestRP2040/ and bumped this library version to 9.4.1. Please try this example and see if you can get it to run on your board.

The schematic shows that only a single RP2040 UART pin is connected to each TMC2209 so only unidirectional commands will work. Since there are no RX pins the RP2040 cannot receive any information from each TMC2209.

The UART pins are not the usual pins reserved for UART0 and UART1 so you can use SerialPIO instead of SerialUART. SoftwareSerial could work, but SerialPIO should be much faster and is preferable.

You will need to add your own code if you want to use the step and direction pins on the RP2040. My example code simply calls moveAtVelocity to test to see if it can get the motors to move. I do not know how large of motors you are using and your power supply voltage so you may  need to adjust the run current.

fryc88, 2024-07-14:
Thank you for your effort. When I'll get home I'll try to flash and test
it. I almost passed on this board, spent most of day yesterday and some
time today and couldn't get anything to work there.

fryc88, 2024-07-16:
Hi,
I tried to compile above, but again i have serial related error:
[image]

I can't even see any libraries for SerialPIO or SoftwareSerial, but neither of them work. Is there anything else that i need to add to Arduino IDE in order to use above?

fryc88, 2024-07-16:
Ok, so I've tried literally all boards from this library:
https://github.com/earlephilhower/arduino-pico/tree/master

Neither of them do anything, however only this library for 2040 does go through the compilation in Arduino IDE. There are other two rp2040 libraries that I've found available, but neither of them will compile, both showing serial issue.
Now I'm thinking why is that happening, I don't fully understand differences between boards, I would assume chip is chip, and as long as hardware connections are set properly then it should work, but it looks like I'm very wrong. I did ask author of that library for any suggestions, maybe he will have some ideas, as I'm pretty much clueless at this stage.

peterpolidoro, 2024-07-16:
Sounds like the Arduino IDE might not be setup properly. You need to make sure you have the correct board and dependencies.

You can also try using platformio instead of the Arduino IDE. I have an example platformio file at the root of the directory. That should automatically download the dependencies for you.