janelia-arduino/TMC2209 #80: EspDuino+CNCShield can't connect to the TMC2209 via UART

18Markus1984, 2024-08-28:
Hello. I have the problem to probably connect to the uart of the TMC 2209. I have some experiences with Arduino and stepper drivers, but not that much to be able to figure out what my issue is. I used your example for the bidirectional communication to analyze the stallGuard behavior. Sadly, I didn't get the stepper to connect. The plan is to use the stepper as a cable cutting machine and detect if the isolation was cut and stop. I added the pinout plan for the EspDuino on the CNCShield and also added my cabling. And yes I know nobody likes the espduino+CNCShield mix because the pins are mixed up but with the ESP you have 3 Serial ports and can still use the board with some workarounds.

I tried to show my setup as clearly as possible.
[image]

The code that I used 
~~~cpp
#include <TMC2209.h>

TMC2209 cutter;
HardwareSerial & serial_stream = Serial2;

const long SERIAL_BAUD_RATE = 115200;
const int DELAY = 1000;
// current values may need to be reduced to prevent overheating depending on
// specific motor and power supply voltage
const uint8_t RUN_CURRENT_PERCENT = 20;
const int32_t VELOCITY = 20000;
const uint8_t STALL_GUARD_THRESHOLD = 50;
const int RX_PIN = 2;
const int TX_PIN = 4;
const int enablePin = 12;
//CNC-Shield with ESPduino
//#define STEP_X 26
//#define STEP_Y 25
//#define DIR_X 16
//#define DIR_Y 27

void setup()
{
  Serial.begin(115200);
  pinMode(enablePin, OUTPUT);
  digitalWrite(enablePin, LOW);

  cutter.setup(serial_stream, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_0, RX_PIN, TX_PIN);
  cutter.setRunCurrent(RUN_CURRENT_PERCENT);
  cutter.setStallGuardThreshold(STALL_GUARD_THRESHOLD);
  cutter.enable();
  cutter.moveAtVelocity(VELOCITY);
}

void loop()
{
  if (not cutter.isSetupAndCommunicating())
  {
    Serial.println("Stepper driver not setup and communicating!");
    return;
  }

  Serial.print("run_current_percent = ");
  Serial.println(RUN_CURRENT_PERCENT);

  Serial.print("stall_guard_threshold = ");
  Serial.println(STALL_GUARD_THRESHOLD);

  uint16_t stall_guard_result = cutter.getStallGuardResult();
  Serial.print("stall_guard_result = ");
  Serial.println(stall_guard_result);

  Serial.println();
  delay(DELAY);
}

jacycatlin, 2024-08-29:
I can only suggest what helped me, which was trying the TMC2209 configurator software I found at https://learn.watterott.com/silentstepstick/configurator/ - finding out whether this works or not could help you rule out possible causes.  If the tool can communicate with the chip, then the UART mode jumper/resistor/whatever is not the problem.
The next thing I did was to use a second microcontroller to report on the serial signals leaving the computer while using the configurator and compare them to the ones sent by the ESP32 (and in my case, once again that wasn't the problem, but at least I was sure it wasn't the driver chip itself)

18Markus1984, 2024-11-28:
Yes I tried both pins to be sure but did't had any results. I didn't had time in the recent weeks, but read throw some TMC documents for the communication. I also ordered the variant of the TMC2209 with the soldered pins for easier testing. I will add the progress when I'm, getting somewhere.