janelia-arduino/TMC2209 #81: MCU Hangs When Setting Specific Address in TMC2209 Library with STM32G0B1RET6

tfranssen, 2024-12-12:
Hi, 

I am experiencing an issue using the TMC2209 library on an STM32G0B1RET6-based board. The board schematic is available here:
`https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V3.0/Hardware/BTT%20E3%20SKR%20MINI%20V3.0_SCH.pdf`

The board has 4 drivers, each with a unique address configured via the MS1 and MS2 pins. When using a simple setup method, everything works as expected. However, the MCU hangs when I try to specify a specific driver using its address during the setup process. The program never reaches the `Setup done.` output/

### Working Code (Default Address)
This code works correctly:

```
#include <SoftwareSerial.h>
#include <TMC2209.h>

SoftwareSerial SoftSerial(PC11, PC10);
TMC2209 stepper_driver_0;
const TMC2209::SerialAddress SERIAL_ADDRESS_0 = TMC2209::SERIAL_ADDRESS_0;

void setup() {
  Serial.begin(115200);  // Init serial port and set baudrate
  while (!Serial)
    ;  // Wait for serial port to connect
  delay(1000);
  Serial.println("Start...");

  // TMC2209::SERIAL_ADDRESS_0 is used by default if not specified
  stepper_driver_0.setup(SoftSerial);
  delay(1000);
  Serial.println("Setup done.");
}

void loop()
{

}
```

### Non-Working Code (Specific Address)
This code causes the MCU to hang indefinitely:
```
#include <SoftwareSerial.h>
#include <TMC2209.h>

SoftwareSerial SoftSerial(PC11, PC10);
TMC2209 stepper_driver_0;
const TMC2209::SerialAddress SERIAL_ADDRESS_0 = TMC2209::SERIAL_ADDRESS_0;

void setup() {
  Serial.begin(115200);  // Init serial port and set baudrate
  while (!Serial)
    ;  // Wait for serial port to connect
  delay(1000);
  Serial.println("Start...");

  // TMC2209::SERIAL_ADDRESS_0 is used by default if not specified
  stepper_driver_0.setup(SoftSerial, 115200, SERIAL_ADDRESS_0);
  delay(1000);
  Serial.println("Setup done.");
}

void loop()
{

}
```

**Steps to Reproduce:**
- Use the STM32G0B1RET6-based SKR Mini E3 V3.0 board.
- Use the provided non-working code.
- Observe the MCU hanging during setup.

**Expected Behavior:**
The MCU should initialize the TMC2209 driver with the specified serial address and continue to execute the program.

**Additional Context:**
Using the default address (no address specified in setup()) works without issues.
The issue occurs only when specifying the address during initialization.

**Questions:**

- Could this be an issue with the library’s handling of serial address selection?
- Are there specific considerations or known limitations when using the library with STM32 microcontrollers?
- Any guidance or suggestions to debug or resolve this issue would be greatly appreciated.

tfranssen, 2024-12-12:
Turned out I didn't had to use Software serial

```
#include <TMC2209.h>
#include <HardwareSerial.h>

// This example will not work on Arduino boards without HardwareSerial ports,
// such as the Uno, Nano, and Mini.
//
// See this reference for more details:
// https://www.arduino.cc/reference/en/language/functions/communication/serial/

const long SERIAL_BAUD_RATE = 115200;
const int DELAY = 3000;

// Instantiate TMC2209
HardwareSerial Serial2(PC11, PC10);

// Instantiate the two TMC2209
TMC2209 stepper_driver_0;
const TMC2209::SerialAddress SERIAL_ADDRESS_0 = TMC2209::SERIAL_ADDRESS_0;
TMC2209 stepper_driver_1;
const TMC2209::SerialAddress SERIAL_ADDRESS_1 = TMC2209::SERIAL_ADDRESS_1;

void setup()
{
  Serial.begin(SERIAL_BAUD_RATE);
  while(!Serial);
  delay(1000);
  Serial.println("Started!");
  // TMC2209::SERIAL_ADDRESS_0 is used by default if not specified
  stepper_driver_0.setup(Serial2, SERIAL_BAUD_RATE, SERIAL_ADDRESS_0);
  stepper_driver_1.setup(Serial2, SERIAL_BAUD_RATE, SERIAL_ADDRESS_1);
  Serial.println("Stepper driver setup complete!");
}

void loop()
{
  if (stepper_driver_0.isSetupAndCommunicating())
  {
    Serial.println("Stepper driver 0 is setup and communicating!");
    Serial.println("Try turning driver power off to see what happens.");
  }
  else if (stepper_driver_0.isCommunicatingButNotSetup())
  {
    Serial.println("Stepper driver 0 is communicating but not setup!");
    Serial.println("Running setup again...");
    stepper_driver_0.setup(Serial2);
  }
  else
  {
    Serial.println("Stepper driver 1 is not communicating!");
    Serial.println("Try turning driver power on to see what happens.");
  }

  if (stepper_driver_1.isSetupAndCommunicating())
  {
    Serial.println("Stepper driver 1 is setup and communicating!");
    Serial.println("Try turning driver power off to see what happens.");
  }
  else if (stepper_driver_1.isCommunicatingButNotSetup())
  {
    Serial.println("Stepper driver 1 is communicating but not setup!");
    Serial.println("Running setup again...");
    stepper_driver_1.setup(Serial2);
  }
  else
  {
    Serial.println("Stepper driver 1 is not communicating!");
    Serial.println("Try turning driver power on to see what happens.");
  } 

  Serial.println();
  delay(DELAY);

}
```

Outputs

```
12:05:14:115 -> Stepper driver 0 is setup and communicating!
12:05:14:115 -> Try turning driver power off to see what happens.
12:05:14:116 -> Stepper driver 1 is setup and communicating!
12:05:14:116 -> Try turning driver power off to see what happens.
```

peterpolidoro, 2024-12-13:
So are you saying that using a non-default serial address works with hardware serial, but not software serial?