janelia-arduino/TMC2209 #82: UART not working

gcormier, 2024-12-21:
I have a scope hooked up to my ESP32-S3 GPIO5 UART pin.

```
  HardwareSerial serialA(1);
  serialA.begin(115200, SERIAL_8N1, SERIALA_RX, SERIALA_TX);
  while(true)
    serialA.print("Hello");
```

This will produce valid serial output on my scope.

```
  HardwareSerial serialA(1);
  serialA.begin(115200, SERIAL_8N1, SERIALA_RX, SERIALA_TX);

  driverA.setup(serialA);
  delay(100);
  driverA.setRunCurrent(STEPPER_RUN_CURRENT);
  driverA.setHoldCurrent(STEPPER_HOLD_CURRENT);
  driverA.enable();
```

This produces no output on my scope. The pin is held high only.

alastaira, 2024-12-27:
See Section 2 of the Serial Setup instructions. You need to call setup() as follows:

```stepper_driver.setup(serial_stream, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_0, RX_PIN, TX_PIN);```

gcormier, 2024-12-28:
Thank you, that got it working!

I now have crashing that occurs when using an interrupt on DIAG for sensorless homing. I'll try to start commenting out libraries to see which is causing the issue.

closed 2024-12-29 by gcormier