janelia-arduino/TMC2209 #84: Passive braking not working

theepicflyer, 2025-01-03:
Hi there,

I'm trying to set up the freewheel / passive braking feature. I first used the [given example](https://github.com/janelia-arduino/TMC2209/blob/main/examples/UnidirectionalCommunication/Standstill/Standstill.ino), but there is no difference in the braking for the 4 modes. 

To my hand it feels like stronger braking than if I manually short both coils with a jumper. I added reading of the registers to check if the registers are setting correctly but it seems like it is. Is it a configuration issue? Am I using features that don't work together? I'm terribly confused by the TMC2209's various features and am not sure what conflicts with that. Any help is appreciated.

## Test Code
``` C
#include <TMC2209.h>

const long SERIAL_BAUD_RATE = 115200;
const int UART1_RX = 26;  // LilyGo UART1 RX pin
const int UART1_TX = 17;  // LilyGo UART1 TX pin

const int32_t RUN_VELOCITY = 14000; // 36000 steps per period * 2 seconds is 1 revolution.
const int32_t STOP_VELOCITY = 0;

const int DELAY = 4000;
const uint8_t RUN_CURRENT_PERCENT = 100;  
const uint8_t HOLD_CURRENT_STANDSTILL = 0;

HardwareSerial &serial_stream1 = Serial1;
TMC2209 driver1;

void setup() {
    Serial.begin(SERIAL_BAUD_RATE);
    driver1.setup(serial_stream1, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_0, UART1_RX, UART1_TX);

    driver1.setRunCurrent(RUN_CURRENT_PERCENT);
    driver1.setHoldCurrent(HOLD_CURRENT_STANDSTILL);
    // driver1.enableAutomaticCurrentScaling();
    driver1.enableCoolStep();
    driver1.enable();
}

void loop() {
    TMC2209::Settings settings;

    Serial.println("Setting standstill mode = NORMAL");
    driver1.setStandstillMode(driver1.NORMAL);

    Serial.println("Reading:");
    settings= driver1.getSettings();
    Serial.print("settings.standstill_mode = ");
    switch (settings.standstill_mode)
    {
        case TMC2209::NORMAL:
            Serial.println("normal");
            break;
        case TMC2209::FREEWHEELING:
            Serial.println("freewheeling");
            break;
        case TMC2209::STRONG_BRAKING:
            Serial.println("strong_braking");
            break;
        case TMC2209::BRAKING:
            Serial.println("braking");
            break;
    }
    Serial.print("settings.irun_percent = ");
    Serial.println(settings.irun_percent);
    Serial.print("settings.ihold_percent = ");
    Serial.println(settings.ihold_percent);
    Serial.println();
    delay(DELAY);

    Serial.println("Setting standstill mode = FREEWHEELING");
    driver1.setStandstillMode(driver1.FREEWHEELING);

    Serial.println("Reading:");
    settings= driver1.getSettings();
    Serial.print("settings.standstill_mode = ");
    switch (settings.standstill_mode)
    {
        case TMC2209::NORMAL:
            Serial.println("normal");
            break;
        case TMC2209::FREEWHEELING:
            Serial.println("freewheeling");
            break;
        case TMC2209::STRONG_BRAKING:
            Serial.println("strong_braking");
            break;
        case TMC2209::BRAKING:
            Serial.println("braking");
            break;
    }
    Serial.print("settings.irun_percent = ");
    Serial.println(settings.irun_percent);
    Serial.print("settings.ihold_percent = ");
    Serial.println(settings.ihold_percent);
    Serial.println();
    delay(DELAY);

    Serial.println("Setting standstill mode = STRONG_BRAKING");
    driver1.setStandstillMode(driver1.STRONG_BRAKING);

    Serial.println("Reading:");
    settings= driver1.getSettings();
    Serial.print("settings.standstill_mode = ");
    switch (settings.standstill_mode)
    {
        case TMC2209::NORMAL:
            Serial.println("normal");
            break;
        case TMC2209::FREEWHEELING:
            Serial.println("freewheeling");
            break;
        case TMC2209::STRONG_BRAKING:
            Serial.println("strong_braking");
            break;
        case TMC2209::BRAKING:
            Serial.println("braking");
            break;
    }
    Serial.print("settings.irun_percent = ");
    Serial.println(settings.irun_percent);
    Serial.print("settings.ihold_percent = ");
    Serial.println(settings.ihold_percent);
    Serial.println();
    delay(DELAY);

    Serial.println("Setting standstill mode = BRAKING");
    driver1.setStandstillMode(driver1.BRAKING);

    Serial.println("Reading:");
    settings= driver1.getSettings();
    Serial.print("settings.standstill_mode = ");
    switch (settings.standstill_mode)
    {
        case TMC2209::NORMAL:
            Serial.println("normal");
            break;
        case TMC2209::FREEWHEELING:
            Serial.println("freewheeling");
            break;
        case TMC2209::STRONG_BRAKING:
            Serial.println("strong_braking");
            break;
        case TMC2209::BRAKING:
            Serial.println("braking");
            break;
    }
    Serial.print("settings.irun_percent = ");
    Serial.println(settings.irun_percent);
    Serial.print("settings.ihold_percent = ");
    Serial.println(settings.ihold_percent);
    Serial.println();
    delay(DELAY);
}
```

## Output: 
Braking (based on turning by hand), is the same for all 4 states, and is stronger than manually shorting both coils.
```
Setting standstill mode = NORMAL
Reading:
settings.standstill_mode = normal
settings.irun_percent = 100
settings.ihold_percent = 0

Setting standstill mode = FREEWHEELING
Reading:
settings.standstill_mode = freewheeling
settings.irun_percent = 0
settings.ihold_percent = 0

Setting standstill mode = STRONG_BRAKING
Reading:
settings.standstill_mode = strong_braking
settings.irun_percent = 100
settings.ihold_percent = 0

Setting standstill mode = BRAKING
Reading:
settings.standstill_mode = braking
settings.irun_percent = 0
settings.ihold_percent = 0
```

peterpolidoro, 2025-01-03:
Sometimes it is hard to tell much of a difference between the braking modes, depending on the exact motor and setup. You should be able to tell a difference between freewheeling and the other braking modes however. It should be noticeably easier to spin the motor in freewheeling mode than in the other braking modes.

I do not see anything obviously wrong in your code. Have you tried any other examples to see if you can communicate with the motor driver and get your motor to spin properly?

theepicflyer, 2025-01-03:
Hey Peter, thanks for the quick reply. Yes all 4 modes, including freewheeling, feel exactly the same. 

I am currently able to use the `moveAtVelocity` with no problems for my current functionality (no STEP / DIR). I have also tried the `TestCommunication` and `SettingsAndStatus` examples successfully. 

One wrinkle that I am now realising is that I am using [these TMC2209 modules](https://github.com/FYSETC/FYSETC-TMC2209/blob/main/images/2209V3.1%E5%8E%9F%E7%90%86%E5%9B%BE.png) (schematic below) that have the ENN pin pulled high. Does this have an impact? Would I need to

[image]

peterpolidoro, 2025-01-03:
Yes, you need to pull the enable pin low in order for the move and standstill functions to work properly.

theepicflyer, 2025-01-06:
I've made sure the enable pin is pulled low and added a read for the hardwareDisabled() reading. Here is the updated code and the output. All 4 states still feel the same strong-ish braking to the hand.

```C
#include <TMC2209.h>

// LilyGo ESP32 UART configuration
const long SERIAL_BAUD_RATE = 115200;
const int UART1_RX = 26;  // LilyGo UART1 RX pin
const int UART1_TX = 17;  // LilyGo UART1 TX pin

// Motor parameters
const int32_t RUN_VELOCITY = 14000; // 36000 steps per period * 2 seconds is 1 revolution.
const int32_t STOP_VELOCITY = 0;

// Test specific parameters
const int DELAY = 4000;
const uint8_t RUN_CURRENT_PERCENT = 100;  
const uint8_t HOLD_CURRENT_STANDSTILL = 0;

// Init global variables
HardwareSerial &serial_stream1 = Serial1;
TMC2209 driver1;

void setup() {
    // Initialize Serial for debugging
    Serial.begin(SERIAL_BAUD_RATE);
    
    // Setup TMC2209 
    driver1.setup(serial_stream1, SERIAL_BAUD_RATE, TMC2209::SERIAL_ADDRESS_0, UART1_RX, UART1_TX);
    driver1.setRunCurrent(RUN_CURRENT_PERCENT);
    driver1.setHoldCurrent(HOLD_CURRENT_STANDSTILL);
    // driver1.enableAutomaticCurrentScaling();
    driver1.enableCoolStep();
    driver1.enable();
}

void loop() {
    
    change_mode(TMC2209::NORMAL);
    driver1.moveAtVelocity(RUN_VELOCITY);
    delay(DELAY);
    driver1.moveAtVelocity(STOP_VELOCITY);
    delay(DELAY);

    change_mode(TMC2209::FREEWHEELING);
    driver1.moveAtVelocity(RUN_VELOCITY);
    delay(DELAY);
    driver1.moveAtVelocity(STOP_VELOCITY);
    delay(DELAY);

    change_mode(TMC2209::STRONG_BRAKING);
    driver1.moveAtVelocity(RUN_VELOCITY);
    delay(DELAY);
    driver1.moveAtVelocity(STOP_VELOCITY);
    delay(DELAY);

    change_mode(TMC2209::BRAKING);
    driver1.moveAtVelocity(RUN_VELOCITY);
    delay(DELAY);
    driver1.moveAtVelocity(STOP_VELOCITY);
    delay(DELAY);
}

void change_mode(TMC2209::StandstillMode mode) {
    TMC2209::Settings settings;
    bool hardware_disabled;

    Serial.print("Setting Mode: ");
    switch (mode)
    {
        case TMC2209::NORMAL:
            Serial.println("normal");
            break;
        case TMC2209::FREEWHEELING:
            Serial.println("freewheeling");
            break;
        case TMC2209::STRONG_BRAKING:
            Serial.println("strong_braking");
            break;
        case TMC2209::BRAKING:
            Serial.println("braking");
            break;
    }
    
    driver1.setStandstillMode(mode);

    Serial.println("Reading:");
    settings = driver1.getSettings();
    hardware_disabled = driver1.hardwareDisabled();
    Serial.print("hardware_disabled = ");
    Serial.println(hardware_disabled);
    Serial.print("Mode = ");
    switch (settings.standstill_mode)
    {
        case TMC2209::NORMAL:
            Serial.println("normal");
            break;
        case TMC2209::FREEWHEELING:
            Serial.println("freewheeling");
            break;
        case TMC2209::STRONG_BRAKING:
            Serial.println("strong_braking");
            break;
        case TMC2209::BRAKING:
            Serial.println("braking");
            break;
    }
    Serial.print("irun_percent = ");
    Serial.println(settings.irun_percent);
    Serial.print("ihold_percent = ");
    Serial.println(settings.ihold_percent);
    Serial.println();
}
```

```
Setting Mode: normal
Reading:
hardware_disabled = 0
Mode = normal
irun_percent = 100
ihold_percent = 0

Setting Mode: freewheeling
Reading:
hardware_disabled = 0
Mode = freewheeling
irun_percent = 100
ihold_percent = 0

Setting Mode: strong_braking
Reading:
hardware_disabled = 0
Mode = strong_braking
irun_percent = 100
ihold_percent = 0

Setting Mode: braking
Reading:
hardware_disabled = 0
Mode = braking
irun_percent = 100
ihold_percent = 0

Setting Mode: normal
Reading:
hardware_disabled = 0
Mode = normal
irun_percent = 100
ihold_percent = 0
```

closed 2025-01-16 by theepicflyer